Ai-Thinker-Open/GPRS_C_SDK #175: Need to hold RST_KEY when powered via VBAT

jussetain, 2018-07-25:
When I power the chip with a 5v battery via VBAT, I need to keep held the RST_KEY for making it working, when I release the RST_KEY, it shuts off.

Is there a solution for that ?

Thanks !

jussetain, 2018-07-26:
Oh, sorry for that, it was a misnomer, I'm actually with a 4.2V battery. 
I found that we must hold the PWR button for 1-2s for booting the module up, is there an alternative way to boot it up? (like a pin or whatever)

gsmwhatnot, 2018-07-26:
For crude and easy way you can connect PWR_KEY ( PIN number 9 ) on module to GND so every time you connect your battery module will power up automatically. I think by doing this RESET pin will not work properly ( That's what I found on bad translated version of Chinese PDF  )

Neutree, 2018-07-28:
that's exactly, you must use 4.2v power supply and  pull down the power key for a while to power on module

jussetain, 2018-07-29:
Thanks a lot for your replies !

closed 2018-07-29 by jussetain