Ai-Thinker-Open/GPRS_C_SDK #178: play audio on call

updesh31, 2018-07-27:
## 1. SDK version（SDK 版本）

{

SW_V2112_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I stored an audio (amr) file in module successfully.How can I play it at the time of call?

}

---

Neutree, 2018-07-28:
can't play audio while making a call

updesh31, 2018-07-28:
At present I am using Mobiletek L206C module in which by AT+CMEDPLAY=1,1.amr,1,100 I can play voice response (not at the time of making call but after answer the call). So I only want to play voice after answering the call.

updesh31, 2018-07-31:
Any function to play saved voice after answering????

Neutree, 2018-07-31:
refer to mp3 play, but parameters different, see the header file (api_audio.h）

updesh31, 2018-07-31:
After start answering I used "AUDIO_Error_t ret = AUDIO_Play(path,AUDIO_TYPE_AMR,PlayCallback);" function which gives error 13 (AUDIO_ERROR_BUSY_CALLING).

Neutree, 2018-08-01:
yes, like most Feature phone, it can't play music(audio) when making a call

updesh31, 2018-08-01:
Is it possible in future?