Ai-Thinker-Open/GPRS_C_SDK #180: <build.bat app> errors

giovannicandotti, 2018-07-28:
## 1. SDK version（SDK 版本）

{

SDK 4.2

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

building app
<code>build.bat app</code>

}

---
## 3. output (output)

{

GEN               stripped (rm syms) elf file app.elf
mips-elf-ld.exe: hex/app/app.elf: access beyond end of merged section (-99877248)

CP                AUDIO CFP FILE......
AUDIO CFG         No audio cfp file
SREC              app.srec for flash/romulator
53
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:543: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2

ROM    total:1048576 Bytes     used:464 Bytes
RAM    total:1048576 Bytes     used:16 Bytes

}

gsmwhatnot, 2018-07-28:
I tried `build app` and it compiled without any problem. I think your `make.exe` ( which executes inside build.bat ) is not pointing to the right make command. Try running `which make` to make sure you are using `make.exe` inside CSDK4.2 folder.
Mine showing this output : `C:\A9_Devel\CSDTK42\make/make.exe`

Please make sure you are using latest CSDK build tools. Here are links : 
CSDK build library :
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases/download/V2.112/GPRS_C_SDK_V2112.7z

CSDK build tools : 
https://mega.nz/#!uKAlRA7Z!UPHn3ZoLY8X33iMIKcBcUYFwrhmNYe-x_dyOdrm_d6I

Please refer to this manual or ask here if you still have problem ...
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html

giovannicandotti, 2018-07-28:
Hi, thanks for support.
I've:
-) removed all CSDTK and GPRS_C_SDK
-) downloaded again, and unzipped, from your links

Results (on Windows10):
-) the script <code>config_env_admin.bat</code> in C:\CSDTK42 directory, executed within <code>cmd.exe</code> shell, does not set environment properly, and PATH does NOT include additional directories
-) on Powershell, the <code>built.bat app</code> fails as before (of course).
-) added additional directories with <code>$Env:path = "C:\CSDTK42\make64;C:\CSDTK42\perl\bin;C:\CSDTK42\python27;C:\CSDTK42\mips-elf-4.4.2\bin;C:\CSDTK42\mips-rda-elf-7.1.0\bin;C:\CSDTK42\rv32-elf-5.4.1\bin;C:\CSDTK42\mingw32\bin;" + $env:path</code> powershell command; set the variable $SOFT_WORKDIR with the powershell command <code>$Env:SOFT_WORKDIR = "C:\GPRS_C_SDK"</code>

Now <code>which make</code> shows properly C:\CSDTK42\make64/make.exe (mind the *64* due to the operating system)

But after all this. still the same error.
To help debugging, let me post a detail: the app_main.c at row 65 is missing a <code>(void **)</code> to cast type; may be I'm not using the proper source files?

gsmwhatnot, 2018-07-28:
No problem at all, I am just a customer like you which happened to have same issues before. I am just helping other to make this product develop in a better way so every one can benefit.

In order to run `config_env_admin.bat` properly you have to run it as administrator or run cmd.exe as administrator then run `config_env_admin.bat` inside cmd.exe

Unfortunately some source codes are not sanitized so I too sometimes use type casting to suppress warning, etc. Still I am suspicious about your environment path. Please run `config_env_admin.bat` again with administrator right and try again ( don't forget to restart windows after this ) Also please test whether you are able to compile other programs inside demo directory or not. Like `build demo call` or `build demo ussd`

I will be online here for the next hour so if you still have problem let me know, maybe we can figure it out together or with remote desktop. I prefer "Anydesk" program for remote desktop.

gsmwhatnot, 2018-07-28:
Also I prefer having 32bit windows for developing. I installed Windows 7 x86 inside a virtual machine and I am developing everything inside it. I think I had some problems with compiling like you around 5-6 months ago ...

updesh31, 2018-07-31:
just install visual c++ redistributable 2010 and 2015 problem will resolved.

giovannicandotti, 2018-08-02:
problem solved!
the "coolwatcher" directory is needed in the PATH variable, from the CSTDK directory.

closed 2018-08-02 by giovannicandotti