Ai-Thinker-Open/GPRS_C_SDK #182: GPS langitude and longitude does not change

Tanukaroy4494, 2018-07-31:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

[image]

}

---

Neutree, 2018-07-31:
not fix yet according to you pic, check you antenna，stick a picture here about your dev board and connection of antenna

Tanukaroy4494, 2018-07-31:
Hi, I am using A9G development board, trying to code using CSDK. 
[image]

USB TX->HST_RX
USB RX-> HST_TX
External VCC->VUSB
External GND->GND

punee995, 2018-07-31:
try testing it on open area or near window!

punee995, 2018-07-31:
of if you have active antenna then it woks like charm.

Tanukaroy4494, 2018-07-31:
Problem was with the power supply , it was not getting enough. Thank you.

closed 2018-07-31 by Tanukaroy4494

Marklanet, 2022-05-30:
how much power supply did you give for it to function well...the voltage and current