Ai-Thinker-Open/GPRS_C_SDK #187: "FS_O_APPEND" doesn't work. The file remains empty!!

Bhavikjp, 2018-08-02:
Hello,
I have used A9G GPS/GPRS module.
Now, I want to save/write/store multiple GPS data into the same file on the memory card/TF card.
**FS_O_RDWR** function works very well. But it overwrites data every time.
I have merged code and used **FS_O_APPEND** instead of **FS_O_RDWR** for appending data into the same file.
**but APPEND function doesn't work. It stores nothing in the file.**

Code for storing data is like this:

[image]

**Can you please suggest a way to solve it out**

Thanks!!

punee995, 2018-08-02:
you're doing it wrong! you still need write to file so.
 
**FS_O_CREATE | FS_O_RDWR | FS_O_APPEND**

Bhavikjp, 2018-08-02:
Hi @punee995 , the thing is i want to add data to one file itself but file carries only one data. Hence  append is not working. Can you suggest some other way. Help would be appreciated.

punee995, 2018-08-02:
did you tried my last comment!

Bhavikjp, 2018-08-02:
@punee995, Yes i did.

punee995, 2018-08-03:
this is the only way to append file as i told you earlier and i've tested it.

`FS_O_CREATE | FS_O_RDWR | FS_O_APPEND`

Bhavikjp, 2018-08-03:
FS_O_CREATE | FS_O_RDWR | FS_O_APPEND
It works now. 
Thanks @punee995

closed 2018-08-03 by Bhavikjp