Ai-Thinker-Open/GPRS_C_SDK #190: GPS低功耗状态下关不掉

Joker-Zhao, 2018-08-05:
## 1. SDK version（SDK 版本）

{

V2.112
V2.112
V2.112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

    GPS_SetLpMode(9);
    GPS_SetStandbyMode(0);
    GPS_Close();            
    UART_Close(UART_GPS);  

//关闭GPS，尝试过每种单独方法以及组合方法，关不掉，低功耗模式下还是有30毫安左右电流，在1xxx版本能关掉

}

---

punee995, 2018-08-05:
you forget about RDA is clocked at 312MHz try setting it at lower freq. or put it in sleep mode.
see Power Management API [here](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_hal_pm.h)

Joker-Zhao, 2018-08-05:
GPS_Close();          
    UART_Close(UART_GPS);   

    OS_Sleep(20);

    Trace(1,"Sleep now");
    PM_SleepMode(true);   

这是我的代码，没有什么作用
This is my code

Neutree, 2018-08-06:
GPS_Close()是关闭GPS的电源，调用后肯定是会有明显电流降低的，应该会降低30mA+，
是不是调用PM_SleepMode进入低功耗后电流没有降低呢