Ai-Thinker-Open/GPRS_C_SDK #192: Typo's and missing documentation for the time API.

mrgreywater, 2018-08-07:
## 1. SDK version（SDK 版本）
V2.112
---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
Time API.

The time API seems weird and not correctly documented:
- [https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/std_inc/time.h#L15](Typo) `TIME_GetIime` should be `TIME_GetTime`
The time returned seems to have an offset of 946684800 from epoch, so it returns the time in seconds from 1/1/2000 12:00AM UTC. Maybe this info could be added to the documentation.
- https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/std_inc/time.h#L16(Typo)
`TIME_GetRtcTIme` should be `TIME_GetRtcTime`
- The time from `time(NULL)` seems to be adjusted to local time. Officially, it should not be locally adjusted. Otherwise it appears to be in seconds from epoch.
---

Neutree, 2018-08-16:
Thanks!

That's right the `TIME_GetTime` return seconds from  2000-1-1 00:00:00 UTC.

And `time(NULL)` return seconds from 1970-1-1 00:00:00 CST(China Standard Time UTC+8:00), it will be changed  to UTC the next version

Neutree, 2018-08-16:
I made some update here: https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/time/src/demo_time.c

closed 2018-09-04 by mrgreywater