Ai-Thinker-Open/GPRS_C_SDK #195: Trigonometric functions not implemented

mrgreywater, 2018-08-10:
## 1. SDK version（SDK 版本）

V2.112

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
Trigonometric functions are declared in "math.h" but are not actually defined in the library. Therefore it will not compile.

```cpp
#include "stdint.h"
#include "stdbool.h"
#include "stdlib.h"
#include "stdio.h"
#include "api_debug.h"
#include "api_os.h"
#include "api_event.h"
#include "math.h"
#include "sdk_init.h"
#include "limits.h"
#include "float.h"
#include "time.h"

void std_Main()
{
    Trace(1,"sqrt:%f", sqrt((double)clock()));
    Trace(1,"sin:%f", sin((double)clock()));
    Trace(1,"cos:%f", cos((double)clock()));
}
```
will result in the following error:
```
error.c:32: relocation truncated to fit: R_MIPS16_26 against `sin'
error.c:33: undefined reference to `sin'
value:0, p:ffffffff88240932, addend:0, symbol:0
error.c:33: relocation truncated to fit: R_MIPS16_26 against `sin'
error:34: undefined reference to `cos'
value:0, p:ffffffff8824094a, addend:0, symbol:0
error:34: relocation truncated to fit: R_MIPS16_26 against `cos'
error.c:34: undefined reference to `cos'
value:0, p:ffffffff88240962, addend:0, symbol:0
error.c:34: relocation truncated to fit: R_MIPS16_26 against `cos'
error:35: undefined reference to `sqrt'
value:0, p:ffffffff882409c0, addend:0, symbol:0
error:35: relocation truncated to fit: R_MIPS16_26 against `sqrt'
error: undefined reference to `sqrt'
value:0, p:ffffffff882409da, addend:0, symbol:0
error.c:35: relocation truncated to fit: R_MIPS16_26 against `sqrt'
```

---

mrgreywater, 2018-08-11:
Yes, it's the latest github (recursive) clone with the SDTK 4.2. But i will retry on a different clean machine soon, just to be sure.

gsmwhatnot, 2018-08-24:
Like I said in posts before, this is caused by some missing CFLAG option of -lm ( -LM ) in build rules. I tried adding it but it didn't work so I asked Neutree to look into it.

My post : https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/169#issuecomment-408635957

Neutree, 2018-08-25:
OK, I'll check it

Neutree, 2018-08-27:
@mrgreywater @gsmwhatnot  
I made some fix, now you can use the variable in the math function!

closed 2018-09-04 by mrgreywater

reallyczx, 2019-11-22:
I'm having the same issues.how am I to solve it?

reallyczx, 2019-11-22:
这个问题是怎样解决的，我使用的是V2.112，问题同样存在？

reallyczx, 2019-11-22:
这个问题修复发布在那个版本里？