Ai-Thinker-Open/GPRS_C_SDK #198: Firmware not updating using demo/FOTA

Tanukaroy4494, 2018-08-11:
## 1. SDK version（SDK 版本）

{

.108

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I am trying to use FOTA but it doesnot get upgraded.Why?
[image]

I am getting response code 200 and proper content length.Immeadiate help will be appreciated.Thanks in advance.

}

---

punee995, 2018-08-24:
did you follow this?
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/fota.html

Tanukaroy4494, 2018-08-24:
@punee995 Yes i did, I am getting proper content length but data isn't as given in my query. Besides response for **API_FotaReceiveData(unsigned char *data, int len)**  is **0**.

punee995, 2018-08-24:
same as #133.
I say read doc carefully.

Tanukaroy4494, 2018-08-24:
@punee995 @Neutree If it encounters space in .pack file , it does not read data further.

punee995, 2018-08-24:
it bad manner to use space in URL.
but if you want to use space then you can simply replace space with **"%20"**.
e.g. file **"V3.0 to new.pack"** will be http://xyx.com/V3.0%20to%20new.pack

Tanukaroy4494, 2018-08-24:
@punee995 my url is **http://api.atozinfoway.in/Prod/IOT/%stonew.pack**  Data inside is
**RDAOTA01T±´}     $ Á[/î¨  ]   d$       0 ¾33” Xè%§¸LÀ
[ÊÉV~íC“N÷·'æ’$!òX’;/ÞÂ7	ûv?–i†@>>Îõ,¸¤I4"C|‘WR«”qhìe–w11Ì–	¯-kSÇÁåà:ØøZGw@2ž&FŽ,çù;ú
~ç!Q±FÓ©`Z0äÊ+¤åÂ¸	×ÈÇ™¤Áw÷S\ÿ1>2þIÂ–¡Mz~4ÆVŸ‰ætè½®•nK´2¢È±‚ÌêÙXÄšH²#h\²Çð$^˜>zlÍ.Ïs%¹Úd#CQ1xi}•èØáA£ëp^*Þ{	¸Ò¬×
?vºØœ”*[ìœŸ‘Bß”Ki”®¢k{¥sž]’ú¯2ëé•¡Ê8û*Ï±måqEÏïgï='‡îÀkµM{>ŠrÎ‹KÓèsù¥æeÂ#Ö"Lõ±Õ+ìâùÈ µÍm)>KfBÿ“øqxYøÍÿ•(F©ü|Þûš0.VÀ…óƒÀeÄ%Søõ‘61¥°îoÁpMGÑ‘ª­`ºÎ±'\Y†éfRX¾év¬Yäå[ùÇÚ­üûR+tÍ[ BùÝS=ø)d	;€Ë*lßµ;ðÄ½._ª>KfBÿ“øqxYøÍÿ•(F©ü|Þûš0.VÀ…óƒÀeÄ%Søõ‘61¥°îoÁpMGÑ‘ª­`ºÎ±'\Y†éfRX¾év¬Yäå[ùÇÚ­üûR+tÍ[ BùÝS=ø)d	;€Ë*lßµ;ðÄ½._ª>KfBÿ“øqxYøÍÿ•(F©ü|Þûš0.VÀ…óƒÀeÄ%Søõ‘61¥°îoÁpMGÑþAe** 

It only takes data till  "RDAOTA01T±´}". Why?

punee995, 2018-08-24:
my output (Working!)
[image]

Tanukaroy4494, 2018-08-24:
@punee995  Can power be the issue?

punee995, 2018-08-24:
mailed working lod files.

Tanukaroy4494, 2018-08-24:
Thank you so much

closed 2018-08-24 by Tanukaroy4494

will88china, 2019-04-12:
@punee995  I have also worked not well,  even the reference code is also abnormal,could you tell me how solve it?thanks.
[image]

will88china, 2019-04-12:
@punee995  could you mail me the working log files too? 172980751@qq.com, thanks so much.

Tanukaroy4494, 2019-04-19:
@will88china Make sure your device has the same firmware build which you used while creating new firmware

will88china, 2019-04-21:
@Tanukaroy4494 yes, according to the online SDK,  I have just modified the version number between old and new firmware, but it always got "fota fail" after  network activate success, I have tried demo code both from SDK  or punee995, but  got same fails, thanks.

LubomirBogdanov, 2019-09-25:
I also experience the same problem. it appears that this function fails:

API_FotaReceiveData( )

in processFotaUpgradeData( ). I also cannot understand what is the purpose of having two defines:

#define SOFT_VERSION          "V1.0.0"
//#define SOFT_VERSION          "V2.0.0"
#define FOTA_HTTP_SERVER      "http://mysite/%sfota.pack"

instead of:

#define SOFT_VERSION          "V1.0.0"
#define FOTA_HTTP_SERVER      "http://mysite/V2.0.0fota.pack"

MaxBoyman, 2019-12-22:
whats the problem with FOTA i exactly did what it said in the [FOTA section of document](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/fota.html) it doesn't work.
[image]
what should i do with this.
i will say what i found from the documentation briefly:
1- cleaning hex folder by "./build.bat clean"
2- compiling code with "./build.bat demo fota" without considering that if it is a realese or debug version
3- back up the hex/fota folder somewhere else and copy fota_*_debug_ota.lod to GPRS_CSDK/ then rename it to old.lod
4- changing whatever you want to provide after program update. for example changing version by modifying SOFT_VERSION from "V3.0" to "V3.1".
5-cleaning hex/fota directory and compiling again without considering that if it is a realese or debug version
6-copy fota_*_debug_ota.lod to GPRS_CSDK/  and rename it to new.lod
7- executing batch with this code: ./build.bat fota old.lod new.lod V3.0tonew.pack
8- uploading V3.0tonew.pack to the host you provided it's link in code.
9- uploading code to A9 or A9G and waiting for update.
I did this procedure with no success. First i used an https server but then i changed the server security to http. again with no success.