Ai-Thinker-Open/GPRS_C_SDK #200: IMEI = 0

ivdor, 2018-08-13:
## 1. SDK version（SDK 版本）

2.1.1.2
CSDTK4.2
---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

1.
First time I saw it on trace of files "demo/gps_agps" or "demo/gps_tracker".
Trace "buffer" before sending DATA to host:

> MMI 02 : send request:POST /?id=000000000000000&timestamp=1483236331&lat=57.817367&lon=28.300472&speed=1.000000&bearing=0.0&altitude=0.000000&accuracy=0.0&batt=79.0 HTTP/1.1

First parameter('id') must be IMEI.

2. 
Next I test file "demo/imei (+iccid)". Trace:

> ```
> [02:16:25.658]          : Fn 2259309 T1 1703 T2 13 T3 09 Time 00 00 47 274
> [02:16:25.658]   MMI 01 : ICCID: 89701200*****1131976
> [02:16:25.763]   MMI 01 : IMEI: 000000000000000
> [02:16:28.658]          : Fn 2259972 T1 1704 T2 00 T3 09 Time 00 00 50 274
> [02:16:28.659]   MMI 01 : ICCID: 89701200*****1131976
> [02:16:28.760]   MMI 01 : IMEI: 000000000000000
> ```

Also imei_build.log has text:

> `Current OS is MSWin32
> mips-elf-ld.exe: hex/imei/imei.elf: access beyond end of merged section (-99877344)`

3.
`AT+CGSN?` command on `8955_modem_v1_AT_release_merge(2).afw` show real IMEI same as qr-code on a9g case

---

ivdor, 2018-08-17:
But CGSN write same imei, as printed on case of module.

I must switch to ICCID or IMSI as unique identifier?
May be i can see imei in memory of module throught tracer?

Neutree, 2018-08-17:
What is the order of this two operation?

1:get imei by AT command
2:get imei by CSDK API

`1 -> 2` or `2->1` or `1->2->1`

ivdor, 2018-08-17:
Doesn't matter. It is different firmwares - original a9g (latest ~2018-06-22) and builded by sdk - demo/imei

AT firmware - show imei.
Builded .lod in sdk - shows "0000000"...

lamqhoang, 2019-03-06:
I have the same problem IMEI show "00000000000000"

marcosasilvalepe, 2020-12-07:
I too have the same problem. Imei shows "00000000000000". Is there a solution to this ?

lamqhoang, 2020-12-08:
My Solution is to write the custom IMEI with AT command AT+EGMR.

ahmedwahdan, 2021-01-24:
Any solution to the empty IMEI, I depend on it for differentiate between device!

marcosasilvalepe, 2021-01-24:
I solved it the same way lamqhoang did. Flashed original AT firmware and with  AiThinker Serial Tool  I did AT command AT+EGMR.

Works OK.

ahmedwahdan, 2021-01-24:
So you write your IMEI?
I have another problem with the serial tool. No baud rate is working. I tried all standard baud rate and I only got garbage on the screen.

marcosasilvalepe, 2021-01-24:
Yes, you have to write the IMEI yourself. I had the number stored before I had this problem so I just wrote the same IMEI.

Regarding the problem with Serial Tool, check where the RX and TX cables are connected in the module. That happened to me as well at some point. 

I don't remember where you're supposed to plug the cables in to work with the Serial Tool. If they're connected to HST_TX and HST_RX try connecting them to RX1 and TX1 and it should work.

ahmedwahdan, 2021-01-24:
Thank a lot.
Regarding this IMEI, but this is not a good way for production handling.
Another point, I'm developing using CSDK, sometimes the application doesn't start, another time after compiling new changes and flashing I find the old App is working not the new one. Any suggestions?

marcosasilvalepe, 2021-01-24:
I didn't played with the CSDK that much. I switched to pulkin's micropython version since it was easier for me to program as I don't have any experience with C.

Maybe someone else could give you some help with that.

ahmedwahdan, 2021-01-24:
Is this micropython stable/well documented?
Any useful links/tutorials?

marcosasilvalepe, 2021-01-25:
micropython is awesome but not all functionality is available on this port, but there is enough to give it a shot.

The link is https://github.com/pulkin/micropython/tree/master/ports/gprs_a9