Ai-Thinker-Open/GPRS_C_SDK #201: SMS_DeleteMessage Doesn't Work!!!

Bhavikjp, 2018-08-19:
**Hello,**

index=5 in SMS_delete function. Stored SMS=4, SMS delete success. But didn't get deleted.
**how to delete all SMS at the same time. as well as one by one.**

[image]

[image]

[image]

**Actually what index does in that function?**

**Thanks!!**

Bhavikjp, 2018-08-21:
@Neutree @punee995 
how to delete all SMS at the same time. as well as one by one.

**Thanks!!**

punee995, 2018-08-22:
I think you're getting it wrong!
if **index_value=5** then it delete **5th message** will be deleted.

i think A9G can store up to ~48 message.
to delete all message you can run **for loop** for 48 times.

Bhavikjp, 2018-08-24:
Yes.later on i have used for loop and it works well.
Thanks for replying!!

closed 2018-08-24 by Bhavikjp

locorcor, 2021-02-05:
Hi Bhavikjp, could you post an example of the loop?  thank you in advance!!!

elangovan27sri, 2022-11-15:
> Hi Bhavikjp, could you post an example of the loop? thank you in advance!!!

hope he used to loop on the sim card that stored messages, not with flash memory.. delete message function on flash is having a bug I believe.