Ai-Thinker-Open/GPRS_C_SDK #204: GPRS Fail after some time being disconnected.

brahmatzadeh, 2018-08-27:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
When GPRS failed due to lack of GSM signal, It won't re-connect to GPRS. When module resets, it connects to network and updates location on ss.neucrack.com and my own website. I tried to reset module (PM_Restart()) each 10 times it sends data to website, the problem solved, but this is a bad idea.

}

---

updesh31, 2018-08-27:
I am also facing same problem.

Neutree, 2018-08-27:
Can the `API_EVENT_ID_NETWORK_REGISTERED_HOME` or `API_EVENT_ID_NETWORK_REGISTERED_ROAMING` come up after the signal is normal?

brahmatzadeh, 2018-08-27:
@Neuyree
Yes, it is handled in the events dispatch routine, with exactly the same code, only changed APN name. 
I used the demo gps_tracker code in C_SDK

brahmatzadeh, 2018-08-30:
Update:
In AT Command there is no problem reconnecting after a while being disconnected from service provider. Anybody found a way to solve the problem?

updesh31, 2018-08-30:
Still not getting success.......

eshlon, 2018-09-02:
I tried to solve similar problem by attempting to reattach and reactivate after API_EVENT_ID_NETWORK_DETACHED ,its for some situations as some special urban areas which there is GSM signal but GPRS going failed  and actually i don't know if there is a better way to trying to attach and activate except a timer as i do ! although the question is what the module do in case of no network situations? I mean its searching for network continuously until it founds something? or a mechanism has been considered ?

eshlon, 2018-09-02:
@Neutree I think its worthy to consider some real life GSM/GPRS scenarios in demo`s sources specially for Tracking applications.

eshlon, 2018-09-03:
It seems that somethings going wrong with http_post function in demo_gps_tracker after being out of network for a long time even after returning back to network and getting attached and activated ; "socket(AF_INET, SOCK_STREAM, IPPROTO_TCP) " returning fd = -1  and wouldn't fixed again .I restart the module after this situation happens for 3 times, but looking for better solution , has  anyone any idea ?

updesh31, 2018-10-20:
?????