Ai-Thinker-Open/GPRS_C_SDK #211: Does A9G support PWM  for led fading ?

TridentTD, 2018-09-01:
Is there any example for led fading ?

Thank you.

gsmwhatnot, 2018-09-02:
There is no PWM pin on A9/A9G module but PWM in nature is nothing but a GPIO pin which goes HIGH/LOW very fast in order to control output voltage. There is `OS_SleepUs` (sleep in microseconds ) function which you can use in order to mimic PWM behavior.

I don't know what is the GPIO voltage of A9 but let's say it's 1.8v so in 1000 microseconds if GPIO goes HIGH for 500 microsecond and goes LOW for another 500 microsecond the voltage reading should be 0.9v ( with ordinary multimeter of-course ) and if it goes 200 microsecond HIGH and 800 microsecond LOW the it should produce 0.36v and so on ...

PS: 
1) This is not a proper PWM approach because PWM cycling frequency is much higher ( +100kHz ) but this should be enough for our eyes to "FEEL" that LED is fading.

2) You should make this function to work parallel with your main code ( I call it pseudo-multi-thread ) otherwise your main code will loop inside PWM indefinitely.

TridentTD, 2018-09-02:
Is there any timer interrupt for creating pin-output frequency in A9G ?

gsmwhatnot, 2018-09-02:
I haven't dove into that area yet, I think you should wait for a proper response from Neutree. Meanwhile you can test to see whether my suggestion works or not, it would be nice to have feedback in here as well.