Ai-Thinker-Open/GPRS_C_SDK #214: Set IMEI within SDK

delimawi, 2018-09-10:
## 1. SDK version（SDK 版本）

{

V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Hi,
In my country, using unsigned IMEI's are prohibited. So I need to assign a known IMEI to the module. In SDK mode how can I change the IMEI?

Thanks

}

---

dewinsmg, 2018-09-28:
AT+EGMR=1,7,"867959033001149"

mini
AT+EGMR=1,7,"867959033XxxXxx"
+EGMR
AT+EGMR=2,7
Before you must download and write the firmware and after send the commands with  serial tool

belalmc, 2019-08-28:
Same problem, @dewinsmg  this is possible with the default firmware using AT commands, but once you compile and upload you can't use AT commands any more,