Ai-Thinker-Open/GPRS_C_SDK #217: Failed to run git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive

yhyuan, 2018-09-13:
## 1. SDK version（SDK 版本）

{
All versions

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
run git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
I will get an error.

Submodule path 'libs/aliyun/iotkit-embedded': checked out 'd096d475cf382bebfa75af36a2d58a0b83c0e477'
fatal: No url found for submodule path 'libs/aliyun/iotkit-embedded/src/packages/Link-CMP' in .gitmodules
Submodule path 'libs/gps/minmea/src': checked out 'cd27e72970cd50b46e9c95c1c0076a49c1ceaf33'
Submodule path 'platform/csdk': checked out 'd6ea87034e2809de154f2014dea031d4559b8a7c'
Failed to recurse into submodule path 'libs/aliyun/iotkit-embedded'

}

---

Neutree, 2018-09-13:
We have to use `git submodule update --init` but `--rescurve` parameter  to clone submodule for aliyun submodule's problem!!!!!

as fllows:

[image]

lsaavedr, 2018-11-21:
This module must be update to [RELEASED_V2_1_20180320](https://github.com/aliyun/iotkit-embedded/tree/RELEASED_V2_1_20180320) branch ?