Ai-Thinker-Open/GPRS_C_SDK #218: How to compile C++ Inheritance  ( Base Class & Derived Class )

TridentTD, 2018-09-14:
CPP from demo can compiled , but when create C++ Base Class & Derived Class 
like as the following , it cannot be compiled. How to solve this issue ?

```
//-----------------------------------------------------
class BaseClass {
public:
    BaseClass() {}
    virtual size_t showInfo(void) = 0;         // Pure virtual function
};

class DeriveClass : public BaseClass {
public:
    DeriveClass() {}
    size_t showInfo(void) override {          // Override function
        Trace(1,"Hello Derive");
    }
};
//-----------------------------------------------------
```

Thank you.

Neutree, 2018-09-14:
Maybe add `-fno-rtti` flag to g++ can resolve your problem

in `platform/compilation/cust_rules.mk`
```makefile
CXX_SPECIFIC_CFLAGS += -Wno-write-strings \
                       -fno-rtti 
```

then

```c
class A{

public:
    virtual void Show() = 0;
    void ShowA()
    {
        printf("class A");
    }
};

class B: public A{

public:
    virtual void Show()
    {
        printf("Show in B");
    }

    void ShowB()
    {
        printf("class B");
    }
};
```

TridentTD, 2018-09-14:
Thank you. It's worked!

TridentTD, 2018-09-14:
@Neutree 

When I create  "write()" or "read()"  virtual function in class,
It occurs some compile-errors.

```
class A{

public:
    virtual void Show() = 0;
    virtual void write() = 0;     // <-------- compile-error ??
    virtual void read() = 0;     // <------- compile-error too ??
    void ShowA()
    {
        printf("class A");
    }
};

class B: public A{

public:
    virtual void Show()
    {
        printf("Show in B");
    }

    void ShowB()
    {
        printf("class B");
    }
};
```

How to solve if I would like to use write() and read() virtual function in a class ?

TridentTD, 2018-09-15:
Compile with C++ Flage -fno-rtti  is ok.
But when create virtual member function name  "write(..)"  or "read(..)",
compilation occures some errors. If I would like to use member function "write()" 
and "read()", how to solve this issue?

```C
class A{
public:
    virtual size_t write() = 0;     // <-------- compile-error ??
    void ShowA()
    {
        printf("class A");
    }
};

class B: public A{
public:    
    virtual size_t write(){
      printf("write in B");
      return 0;
    }
    void ShowB()
    {
        printf("class B");
    }
};
```

Neutree, 2018-09-21:
cause the read and write is defined in api_socket.h

TridentTD, 2018-09-21:
If I  turn off  by  "//#define write  ..." and   "//#define read ..." , 
is there any effect or any problem to other A9G's sdk?