Ai-Thinker-Open/GPRS_C_SDK #219: Can A9G be flashed by command line?

TridentTD, 2018-09-14:
CoolWatcher is great tool to flash firmware to A9G.
However if I would like to flash by command line too,
Is there any way to flash by command line to A9G?

Neutree, 2018-09-14:
no command line tool yet, for we have no download protocol designed by RDA, they just provide us lib file on windows.
there's some ways to make it for us, ask RDA for help, or try to decode the protocol and write our own tool

Neutree, 2018-09-14:
And as you see, the code provided from RDA is open source by some one, you startd it before https://github.com/cherryding1/RDA8955_W17.44_IDH

TridentTD, 2018-09-14:
I have already downloaded RDA89655_W17.44_IDH from the site.
Do you have the web-page to teach about how to build this RDA's  lib files?

Neutree, 2018-09-14:
no doc

TridentTD, 2018-09-14:
Do you mind to descript how to build them?
Thank you .

kaido42, 2018-10-19:
I could write the command-line programmer, not a big problem (investigated the protocol a bit yet), if I could use the a9g in my project. currently the serial-port issue (https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/232) is the biggest showstopper ..