Ai-Thinker-Open/GPRS_C_SDK #220: Maximum voltage at GPIO and ADC

ricardopera, 2018-09-14:
Hi,

I have doubts about how to work safely with the A9G board.

What is the maximum voltage that I can apply to the GPIO and ADC without damaging the board?
Can I damage the board if I apply more than 2.8V to GPIO or more than 1.8V to ADC?
Can I work with sensors and modules that work with 3.3V or 5V with the A9G board?

Thanks.

andyman2240, 2019-04-09:
I noticed that no one has answered  so I thought that I would share my experience with GPIO levels.  I am using a SSD1306 OLED with the OLED Vcc connected to VBat ( approx. 4.1VDC) and SDA/SCL on the OLED connected to GPIO20 and GPIO19 on the A9G. It has been working fine, so I am assuming that the traditional 3.3V sensors will work with the A9G without damaging the A9G.

ricardopera, 2019-04-22:
Thank  you Andyman. I took the risk and tested it with 3.3v sensors and it worked fine too.