Ai-Thinker-Open/GPRS_C_SDK #225: How to establish a connection to a HTTPS endpoint and make POST and GET requests.

cryarchy, 2018-09-18:
SDK version = V2.112
---
Kindly provide me with sample code that sends and/or gets data from an HTTPS endpoint. I have tried to get the SSL example working to no avail. I would have made this request in the a9g forum but I am avoiding the Chinese.
---

Ricx8, 2019-01-21:
Hi @cryarchy I wrote this code that does exactly what you need.

Abibor, 2023-05-07:
@Xoffio Hello, can you help me to make ssl post request to firebase? I successfully make GET request but my post request is fail - "ssl read fail: -6". My post request:
`snprintf(retBuffer,retBufferLen,"POST /GpsTags/.json?auth=mySecurityKey1231232131123 HTTP/1.1\r\nHost: fieldercontrol-11c25-default-rtdb.firebaseio.com\r\nContent-Type: application/json\r\nContent-Length: 31\r\n%s\r\n\r\n", data);`

Xoffio, 2023-05-07:
Hey @Abibor, its being a while since the last time I used this chip and its hard to tell what your problem is but make sure you changed the ssl certificate, the flags and the ssl variables to match with the website you are trying to access.. Go through the entire code and look for my notes. I wrote what you needed to modify