Ai-Thinker-Open/GPRS_C_SDK #227: Coolhost not launching on Linux

cryarchy, 2018-09-19:
OS: Linux Mint 17.3 64-bit
SDK version = V2.112

The problem occurs when I launch **coolwatcher** and it, in turn, tries to launch **coolhost**.
```
./coolhost: relocation error: ./coolhost: symbol _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference
```

Neutree, 2018-09-25:
ubuntu 16.04

vnetmx, 2018-10-01:
You must install this:
sudo apt install build-essential libqt4-qt3support itcl3 itk3 iwidgets4

and try to renaming (like I did to avoid deleting files... just in case) :

CSDTK/lib to CSDTK/lib.tmp

And try again, coolwatcher working in ubuntu 18.04 right now.

cryarchy, 2018-10-02:
Tried it. Same error.

vnetmx, 2018-10-03:
> Tried it. Same error.

Do you have some error message?

try with 'ldd ./coolwatcher' and 'ldd ./libcoolhost_library.so' to know if some lib is missing.

cryarchy, 2018-10-08:
`ldd ./coolwatcher` results:
```
	linux-vdso.so.1 =>  (0x00007fff335a1000)
	libQt3Support.so.4 => /usr/lib/x86_64-linux-gnu/libQt3Support.so.4 (0x00007f1b21e15000)
	libQtNetwork.so.4 => /usr/lib/x86_64-linux-gnu/libQtNetwork.so.4 (0x00007f1b21acd000)
	libQtXml.so.4 => /usr/lib/x86_64-linux-gnu/libQtXml.so.4 (0x00007f1b21888000)
	libcoolruby.so => /home/archy/A9G/CSDTK/cooltools/./libcoolruby.so (0x00007f1b2061e000)
	libcoolxml_library.so => /home/archy/A9G/CSDTK/cooltools/./libcoolxml_library.so (0x00007f1b2036a000)
	libruby.so.1.8 => /home/archy/A9G/CSDTK/cooltools/./libruby.so.1.8 (0x00007f1b20067000)
	liblibelf.so => /home/archy/A9G/CSDTK/cooltools/./liblibelf.so (0x00007f1b1fe48000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1b1fc44000)
	libcoolhost_library.so => /home/archy/A9G/CSDTK/cooltools/./libcoolhost_library.so (0x00007f1b1fa27000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f1b1ed74000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f1b1e88f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1b1e579000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1b1e361000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1b1df98000)
	libQtSql.so.4 => /usr/lib/x86_64-linux-gnu/libQtSql.so.4 (0x00007f1b1dd59000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1b1db3b000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1b1d835000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1b1d61c000)
	libromulib.so => /home/archy/A9G/CSDTK/cooltools/./libromulib.so (0x00007f1b1d416000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1b1d20e000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f1b1cfd5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1b222f2000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f1b1cd99000)
	libaudio.so.2 => /usr/lib/x86_64-linux-gnu/libaudio.so.2 (0x00007f1b1cb81000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f1b1c879000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f1b1c653000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f1b1c3b0000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f1b1c15f000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f1b1bf57000)
	libICE.so.6 => /usr/lib/x86_64-linux-gnu/libICE.so.6 (0x00007f1b1bd3b000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f1b1bb2b000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f1b1b921000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f1b1b70f000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f1b1b3da000)
	libusb-0.1.so.4 => /lib/x86_64-linux-gnu/libusb-0.1.so.4 (0x00007f1b1b1d1000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f1b1afa7000)
	libXt.so.6 => /usr/lib/x86_64-linux-gnu/libXt.so.6 (0x00007f1b1ad41000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f1b1ab3d000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f1b1a8ff000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f1b1a6f7000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f1b1a4f2000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f1b1a2d3000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f1b1a0cd000)
```
`ldd ./libcoolhost_library.so` results:
```
        linux-vdso.so.1 =>  (0x00007fffee5ea000)
	libQtNetwork.so.4 => /usr/lib/x86_64-linux-gnu/libQtNetwork.so.4 (0x00007f95f9bef000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f95f970a000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f95f93f4000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f95f91dc000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f95f8e13000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f95f8bf5000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f95f89dc000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f95f87d8000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f95f84d0000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f95f82c8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f95f7fc2000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f95fa154000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f95f7d84000)
```

lsaavedr, 2018-11-21:
Hello @cryarchy you can comment LD_LIBRARY_PATH settings in your "$HOME/.bashrc" configuration file, then add this in in "build.sh" file to compile... hope this can help you... regards!

Makimars, 2025-02-10:
No one is probably doing this anymore, but I got it working on an up-to-date Fedora and I assume it would've worked on other modern distros as well. I created a VM with Ubuntu 16.04.7, got the program working there and then created an appimage which works on newer distros too. I will probably fork this repo and create a release with that appimage.