Ai-Thinker-Open/GPRS_C_SDK #230: EVENT SNIFFING ERROR: CONNECTION BROKEN?

yhyuan, 2018-09-27:
## 1. SDK version（SDK 版本）

{

aLL SdK.

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
When I tried to burn the new firmware for the first time, I always run in the this error. Sometimes at 91%, sometimes at 94%, sometimes at 1%. 
Here is the message. 

> include CoolWatcher
> $LOAD_PATH << "./rbbase"
> $LOAD_PATH << "./rbbase/common"
> $LOAD_PATH << "C:/a9g/CSDTK42/cooltools/profiles"
> load 'CoolWatcher/cwuser.rb'
[Project set to 8809 for [CT8851]]
Loading common plugin BufferProfile...
Loading common plugin disassembly...
Loading common plugin flamulator...
Loading common plugin keypad...
Loading common plugin tracer...
Loading plugin Calib...
Loading plugin cwmodem2ggui...
Loading plugin fastpf...
Loading plugin romburn...
Opening Chip on COM3...

￼ Connection $CURRENTCONNECTION (COM3) opened (#CURRENTCONNECTION). 

￼ Connection RegWatch Connection (COM3) opened. 

￼ Connection Event Sniffer (COM3) opened. 

￼ Connection BufWatch Connection (COM3) opened. 

Done.

￼ [COM OPEN OK] 

Event sniffer (re)starting. (Connection: Event Sniffer (COM3))
Loading profile script : 8955.rb
EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.
Detected hardware reset (0xffffffff).
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
gps_agps_B2120_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...

Resetting SPI flash...
Done.

Entered Host Monitor mode.

No boot sector struct exists. Skipping ...

0.000000 0.000000 0.000000 ( 0.031498)

Verify enabled: true
Fastpf Protocol Version: 1.5
Got FPC buffer size: 65536
Use FPC buffer size: 32768
Boot sector address: 0x0
Fastpfing...
Event sniffer (re)starting. (Connection:
Event Sniffer (COM3))

￼ `waitCommandResult': Event aknowledging end of command never received! (FastPf::FlashCommandUnacknowledged)

./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
61
`waitCommandResult'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
309
`dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
306
`each'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
306
`dofastpfXfer'
C:/a9g/CSDTK42/cooltools/lib/ruby/1.8/benchmark.rb
293
`measure'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
305
`dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
718
`dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
788
`fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
83
`fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
112
`fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

}

---

yhyuan, 2018-10-02:
Fixed by replacing with a new USB to UART cable.

closed 2018-10-02 by yhyuan

wavresangam, 2018-12-18:
hey yhyuan ,
 i am also facing this problem currently am using cp340 usb to ttl converter suggest me to which usb to ttl converter is  better to use