Ai-Thinker-Open/GPRS_C_SDK #235: Can I use UART1 to send/receive AT commands?

ubaldus, 2018-10-05:
## 1. SDK version（SDK 版本）

{

B2120

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

is there any way to read/write AT commands and GPS data directly?

}

---

kaido42, 2018-10-05:
There is no direct way at the moment.

Should not be very hard to implement, though. Look at other gps-related functions in libs\gps\include\gps.h and their implementations in libs\gps\src\gps.c. You can add a function(s) there .. 

re reading - all what gps is sending is accessible in API_EVENT_ID_GPS_UART_RECEIVED event. Process all you want before calling GPS_Update()

kaido42, 2018-10-05:
oh, I misread your initial question .. if you work with the SDK there is no AT commands any more.
the on-board gps is GOKE, commands for GOKE are described here (did not test them all, though)
http://www.openluat.com/Product/file/530/Air530%20GPS%E6%A8%A1%E5%9D%97%E7%94%A8%E6%88%B7%E6%89%8B%E5%86%8CV1.7.pdf

ubaldus, 2018-10-08:
Hi Kaido, 
it would be very helpful for instance to be notified about incoming calls and being able to establish or receive CSD calls.

kaido42, 2018-10-08:
would be nice to get a event for incoming ring, indeed. at the moment it seems the only way is to poll with CALL_Status() ... Never tried, though .. 

never tried the AT commands ..

ubaldus, 2018-10-08:
Well there is an API_EVENT_ID_CALL_INCOMING which I didn't test yet but I suppose it should be triggered on incoming call, it would be nice if it could also pass the call type (voice/csd/fax/etc) apart the incoming caller id, but still no way to support CSD in/out calls.

nhubach5998, 2018-12-05:
is there any way that I can send AT command from USB port? I used a program to send AT command to TX1 and RX1 before, and it worked fine. After I play around with SDK, I have no clue how to make the chip take my AT command anymore.

nhubach5998, 2018-12-06:
Simply, do you know how to switch back to using AT command via MCU?

punee995, 2018-12-06:
Here you go: https://github.com/Ai-Thinker-Open/GPRS-AT

nhubach5998, 2018-12-06:
Thank you, but after I burn LOD to the chip, AT TX RX port no longer response to my command. 
before I burn anything, AT commands worked just fine.

punee995, 2018-12-06:
Download AT Firmware From Here https://wiki.ai-thinker.com/gprs/firmware

burn it using https://wiki.ai-thinker.com/gprs_download

nhubach5998, 2018-12-07:
Thank you so much, I have been digging for this for couple days now

Rioshop, 2019-07-05:
> is there any way that I can send AT command from USB port? I used a program to send AT command to TX1 and RX1 before, and it worked fine. After I play around with SDK, I have no clue how to make the chip take my AT command anymore.

How did you succeed to send AT commands to TX1 and RX1? Working on it since a few days but it doesn't work...
I use AT commands for a Bluetooth module.

Rioshop, 2019-07-05:
z

Rioshop, 2019-07-05:
e