Ai-Thinker-Open/GPRS_C_SDK #240: PM_SleepMode Question

vnetmx, 2018-10-11:
## 1. SDK version（SDK 版本）

{
2120
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
There is no a problem is a question.
Does "PM_SleepMode" shutdown all peripherals and set this to Min. Frequency? What is the difference with  putting this to PM_SetSysMinFreq(PM_SYS_FREQ_32K) only?
}

---

punee995, 2018-10-11:
This might help -> [pmsleepmode](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/power-manager.html#pmsleepmode)

vnetmx, 2018-10-11:
Thanks @punee995 , already read that but what I didnt understood was this sentence:

> At the same time, it can turn off peripherals and their power to reduce power consumption.

It turn off? or it 'can' but you need to turn off manually?

Sorry, my english is not native one.

punee995, 2018-10-11:
yes peripherals will turn off automatically.

mnorval, 2019-08-16:
I have tested this function where it sleeps for 1 hour (60 * 1 minute OS_Sleeps) and then wakes up.  

_uint8_t tempcount=0;
		PM_SleepMode(true);
		Trace(1,"Enter deep sleep");
		for (tempcount=0; tempcount < 60;tempcount++)
		{
			Trace(1,"Sleep %d",tempcount);OS_Sleep(60 * 1000);//1 minute
		}
		PM_SleepMode(false);
		Trace(1,"Exit deep sleep");_

Battery consumption is still very high.
I suspect GSM remains active.

Is there a command to disable GSM?