Ai-Thinker-Open/GPRS_C_SDK #246: Trying to connect to different MQTT broker than in the demo

dawidjordaan, 2018-10-17:
## 1. SDK version（SDK 版本）

{

CSDTK 4.2 ($Revision: 43828 $)

SDK V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I'm trying to use the MQTT to connect with AWS IOT but cannot connect to the server. No connection is established.

I am using information from this post to setup my MQTT connection settings  https://github.com/Azure/azure-iot-sdk-c/issues/660

I've attached my files i'm using.
[mqtt_ssl_aws.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2489436/mqtt_ssl_aws.zip)

Getting my CA certificate from here 
https://docs.aws.amazon.com/iot/latest/developerguide/managing-device-certs.html

The device certificate and private keys are here:
[Certificate and Keys.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2489450/Certificate.and.Keys.zip)

This is my tracer ouput...
[image]

Sometimes the tracer will also ouput this error after network connection like above.
[image]

}

---

dawidjordaan, 2018-10-24:
This problem was due to my policy linked to the certificate not allowing the connection.

The Qos should also be 0 or 1.

closed 2018-10-24 by dawidjordaan

malte777, 2020-05-17:
Hi,
I also have a problem connecting to AWS:
"MQTT connect to broker fail. Error: 256"
With MQTT-Fx I can connect. But not with the the A9G demo script. I changed the QoS levels from 2 to 1. But it still doesn't work. What else has to be changed?

malte777, 2020-05-18:
I have it running now, by making these changes:
change
    ci.ssl_min_version   = MQTT_SSL_VERSION_SSLv3;
to
    ci.ssl_min_version   = MQTT_SSL_VERSION_TLSv1_2;

delete:
    ci.entropy_custom    = "GPRS_A9";
    ci.will_qos = 2;
    ci.will_topic = "will";
    ci.will_retain = 1;
    memcpy(strstr(willMsg,"GPRS")+5,imei,15);
    ci.will_msg = willMsg;