Ai-Thinker-Open/GPRS_C_SDK #247: C++ operator new missing?

kaido42, 2018-10-18:
in cpp demo, adding

`B *test = new B`

causes the following error during lnking:

`demo\cpp/src/demo_cpp.cpp:79: undefined reference to 'operator new(unsigned int)'`

in another project I was trying to use C++, I'm getting the following:

```
LD                kala.elf
mips-elf-ld: section .gcc_except_table.kalaSoundInit loaded at [ffffffff882490e0,ffffffff882490ef] overlaps section .cust_data loaded at [ffffffff882490e0,ffffffff882490ef]
C:/work/a9g/a9_sdk/build/kala/kala/lib\libkala_debug.a(kala_speaker.o): In function `kalaSoundInit':
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:29: undefined reference to `operator new(unsigned int)'
value:0, p:ffffffff88242860, addend:0, symbol:0
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:29: relocation truncated to fit: R_MIPS16_26 against `operator new(unsigned int)'
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:31: undefined reference to `operator delete(void*)'
value:0, p:ffffffff88242872, addend:0, symbol:0
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:31: relocation truncated to fit: R_MIPS16_26 against `operator delete(void*)'
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:31: undefined reference to `_Unwind_Resume'
value:0, p:ffffffff88242878, addend:0, symbol:0
C:\work\a9g\a9_sdk\kala/src/kala_speaker.cpp:31: relocation truncated to fit: R_MIPS16_26 against `_Unwind_Resume'

```

please help.

piyushverma, 2018-11-26:
I am facing same isssue , any suggestion how to utilise C++ library ?

pergerk, 2018-12-12:
void * operator new(size_t size)
{
  return OS_Malloc(size);
}

void operator delete(void * ptr)
{
  OS_Free(ptr);
}
 You can use this functions. I got new error with unwind_value