Ai-Thinker-Open/GPRS_C_SDK #249: Unable to download any on pundit a9g

thefaisalhaque, 2018-10-21:
I'm not able to download any new firmware. My coolwatcher (2.01.03) stops on this error, after a workig first connection:

￼ `CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

./rbbase/common/coolhost_connections.rb
161
`CHST_Read32'
./rbbase/common/coolhost_connections.rb
161
`read32'
./rbbase/common/base.rb
85
`read'
./rbbase/common/base.rb
239
`read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb
72
`enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb
366
`session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
12
`doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
62
`ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
212
`dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
718
`dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
788
`fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
83
`fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
112
`fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.
-------------------------------------------------------------
SDK version - V2.112 
usb-uart is the FT232RL and the tx->hst-rx and rx->hst-tx pins are connected.

but not working.

dhernandezl, 2019-12-17:
Did you solve it?
I have the same problem

azakh, 2020-01-18:
> Did you solve it?
> I have the same problem

I couldn't download anything as well with CP2102 until I connected
GND->GND, RX->HST_TX, TX->HST_RX and powered the board from USB adapter.
Try connect data pins first and then powerup the board.