Ai-Thinker-Open/GPRS_C_SDK #251: Get integer from json object

Tanukaroy4494, 2018-10-26:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I want to parse json object which has integer as one item
e.g. {"code":101,"url":"www.api"};

I am getting error when parsing this. Also i want to store 101 in integer. How can i ?

}

---

punee995, 2018-10-26:
**aren't this [cJSON ](https://github.com/DaveGamble/cJSON)related issue!**

BTW :-
int code = cJSON_GetObjectItem(root, "code")->valueint;
char *url = cJSON_GetObjectItem(root, "url")->valuestring;

Tanukaroy4494, 2018-10-30:
@punee995 
cJSON_GetObjectItem(root, "code") is of type  cJSON , we cannot assign it to integer.
Anyway i found a solution.

closed 2018-10-30 by Tanukaroy4494

punee995, 2018-10-30:
please follow a simple example [here ](https://spacesciencesoftware.wordpress.com/2013/09/10/a-good-way-to-read-json-with-c/) for the better understanding of cJSON

K1ngM, 2020-04-06:
@Tanukaroy4494 
Hey Tanukaroy4494. Would you mind sharing the solution? 
Or you just wrote integers as characters? e.g. {"code":"101","url":"www.api"};

Tanukaroy4494, 2020-04-08:
Hi, 
You can add integers to json , it is valid.{/"code/":101,/"url/":/"www.api/"}
Please share link of your question.

K1ngM, 2020-04-08:
I found the prolem.. cJSON_Parse uses strtod function which is not properly implemented in A9G "stdlib.h". I fixed cJSON parse library to use atof instead and it works now.

Gruntactic, 2021-07-05:
I ran in the same issue and here is my take on the subject and my work around, thank you for your efforts it has shed a guiding light along my path, it is true that the cJSON_Parse uses strtod function, and the atof function is a valid replacement, but it lacks the pointer portion parameter, which leaves the pointer after_end at it's initial value (NULL), my work around is to increment that variable on each run of the loop just before the switch statement:

`for (i = 0; (i < (sizeof(number_c_string) - 1)) && can_access_at_index(input_buffer, i); i++)`
`    {`
`        after_end ++; // count the number of character to set buffer.offset for the next item`
`        switch (buffer_at_offset(input_buffer)[i])
        {...`

and decrement it once just before assigning it's value to  input_buffer->offset:

`    after_end--; //step back one character to get the coma marker`
`    input_buffer->offset += (size_t)after_end; // set next point`
`return true;`