Ai-Thinker-Open/GPRS_C_SDK #257: GPS not recognizing satellites.

krishanddk, 2018-11-11:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

krishanddk, 2018-11-11:
GPS fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:0, gps sates total:0, is fixed:no fix, coordinate:WGS84, Latitude:22.606252, Longitude:113.840102, unit:degree,altitude:0.000000

This is the value always throws
It was working 3 days before
Now it does not working even with AT commands

krishanddk, 2018-11-12:
My location is in Sri Lanka

punee995, 2018-11-12:
1. Try testing it on open area (open sky or near window).
2. Give proper power supply to module.

krishanddk, 2018-11-16:
I did all that and change the module all the modules are same. Do I have to do anything special?
it give always 
 Latitude:22.606100, Longitude:113.840128

krishanddk, 2018-11-16:
this is the full output 
fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:0, gps sates total:0, is fixed
:no fix, coordinate:WGS84, Latitude:22.606100, Longitude:113.840128, unit:degree,altitude:0.000000

punee995, 2018-11-16:
In the hot start, you have to wait ~1min to get GPS to fix.

nicholaspearson, 2018-11-22:
@punee995 @krishanddk did you get to a solution?

punee995, 2018-11-28:
I'm using V2.112 also I have an active antenna connected to A9g and placed near a window. and get GPS fix(No BDS Fix) around 30 sec.

Ricx8, 2019-01-16:
@krishanddk try this code: https://github.com/Ricx8/GPRS_C_SDK/tree/master/demo/gpsSimpleTest 
Let me know

nicholaspearson, 2019-01-17:
> @krishanddk try this code: https://github.com/Ricx8/GPRS_C_SDK/tree/master/demo/gpsSimpleTest
> Let me know

What's different about this code?

Ricx8, 2019-01-21:
> > @krishanddk try this code: https://github.com/Ricx8/GPRS_C_SDK/tree/master/demo/gpsSimpleTest
> > Let me know
> 
> What's different about this code?

It's just a simple code that only gets the GPS coordinates avoiding other things like try to connect to GSM. That way you can make sure that is the GPS module. It is pretty easy to read the code, check it out if you want to :)

Ricx8, 2019-01-24:
Hi @nicholaspearson and @krishanddk . Today I had the same problem that you have. The code that I give you didn't work so I realize it was the hardware.... I search related issues and I found this one that already exist [Issue #181 ](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/181). I connect my A9G by USB to a power bank and I put it right to the window and works! try something similar I'm pretty sure you have the same issue.

CRONOS-IOS, 2020-05-06:
hola  @nicholaspearson y @krishanddk, el mio estaba trabajando normal, hasta que hoy tuve el mismo problema, intente con otro modulo A9G nuevo y me sale exactamente el mismo problema... alguien lo pudo reparar? o talvez sea un problema con las antenas satelitales?

CRONOS-IOS, 2020-05-06:
hola @nicholaspearson y @krishanddk , el mio estaba trabajando normal, hasta que hoy tuve el mismo problema, intente con otro módulo A9G nuevo y yo venta exactamente el mismo problema ... alguien lo pudo reparar? o talvez sea un problema con los satelites?

CRONOS-IOS, 2020-05-06:
> > @krishanddk prueba este código: https://github.com/Ricx8/GPRS_C_SDK/tree/master/demo/gpsSimpleTest 
> > Avísame
> 
> ¿Qué tiene de diferente este código?

Esto es lo que me sale a mi... 
#LOG: init gps...
#LOG: get gps firmware version fail
#LOG: GPS starting up...
#LOG: (22.607603 N 113.830580 W)

CRONOS-IOS, 2020-05-06:
[17:56:05.594]   SIM 12 : [MAIN_INSTRUCTION]
[17:56:05.594]   SIM 05 : Main instruction: b2
[17:56:05.594]   SIM 12 : Class 0x0, Ins 0xb2, P1=0x14, P2=0x4, P3=0xb0
[17:56:05.609]   CSW 01 : Process nEventId[a0684]\n
[17:56:05.609]          : Fn 0001020 T1 0000 T2 06 T3 00 Time 00 00 37 511
[17:56:05.653]   MMI 01 : #LOG: GPS starting up...
[17:56:05.653]   MMI 01 : #LOG: (22.607603 N 113.830580 W)
[17:56:05.653]   SIM 10 : Int dma success, First RxByte:0xb2 NbRxBytes:179
[17:56:05.653]   SIM 11 : Data received :
[17:56:05.653]   SIM 11 : Dump : at 0xA1C094C9 (176/176) : 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[17:56:05.653]   SIM 12 : [GET_RESPONSE]
[17:56:05.653]   SIM 10 : SW1,SW2=90 0
[17:56:05.653]   SIM 12 : [FETCH]
[17:56:05.653]   SIM 03 : Sim 0 Message Cnf : Sw1=90, Sw2=0
[17:56:05.653]   SIM 03 : Dump : at 0x820C5853 (176/176) : 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[17:56:05.653]   SIM 03 : File :69

martinconstante, 2021-06-03:
I had the same problem, I solved it with an external power supply and put the antenna outside the window.
martin.constante@gmail.com

DarwinVargas2023, 2023-02-25:
AT+CMGF=1 

OK
AT+CMGS=9951548193

> AT+CMGS=9951548193

COMMAND NO RESPONSE!

DarwinVargas2023, 2023-02-25:
That is my problem when i send At command to send sms