Ai-Thinker-Open/GPRS_C_SDK #262: Detected BCPU in the GDB loop (0x9db10000) with international ioT SIM

updesh31, 2018-11-16:
## 1. SDK version（SDK 版本）

{

SW_V2125_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Today I start the A9G with an international ioT SIM card but it crashes at the time of attaching GPRS. Log is attached.
[trace.txt](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2590714/trace.txt)

}

---

PostconEssen, 2019-01-22:
Hello!

Have the same problem here too! I Tried the gps_agps demo this time.
I only changed the APN to "tm" for my Things Mobile SIM. 

```C
Network_PDP_Context_t context = {
                    .apn        ="tm",
                    .userName   = ""    ,
                    .userPasswd = ""
                };
                Network_StartActive(context);
```

Coolwatcher shows this:
Detected BCPU in the GDB loop (0x9db10000), connect GDB debugger for more info.
Detected XCPU in the GDB loop (0x9db00000), connect GDB debugger for more info.

Trace:

```
[17:23:39.044]          : SXS_INIT_RMC

[17:23:40.713]   SXR 01 : Allocated Tasks 19 Schedule Disable 1\n

[17:23:40.713]   SXR 01 : Task  0: Task Idle Priority 255

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/8192 Top 0x820b09f8 <- 0x820b29f8\n

[17:23:40.713]   SXR 01 : Task  1: Layer 1 Priority   1

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/1536 Top 0x820b2a10 <- 0x820b3010\n

[17:23:40.713]   SXR 01 : Task  2: Sim Priority  15

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/800 Top 0x820b3028 <- 0x820b3348\n

[17:23:40.713]   SXR 01 : Task  3: LAPDm Priority  21

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/800 Top 0x820b3360 <- 0x820b3680\n

[17:23:40.713]   SXR 01 : Task  4: RR Priority  25

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/4096 Top 0x820b3698 <- 0x820b4698\n

[17:23:40.713]   SXR 01 : Task  5: RLD Priority  26

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/800 Top 0x820b46b0 <- 0x820b49d0\n

[17:23:40.713]   SXR 01 : Task  6: RLU Priority  27

[17:23:40.713]   SXR 01 : is pending.\n

[17:23:40.713]   SXR 01 : Stack  32/1536 Top 0x820b49e8 <- 0x820b4fe8\n

[17:23:40.713]   SXR 01 : Task  7: LLC Priority  29

[17:23:40.713]   SXR 01 : is pending.\n
[... 283 lines cut ...]
[17:23:49.734]   SXR 01 : 88092665y\n

[17:23:49.734]   SXR 01 : 88184741y\n

[17:23:49.734]   SXR 01 : 88184747y\n

[17:23:49.734]   SXR 01 : 88121b9dy\n

[17:23:49.734]   SXR 01 : 8200df85y\n

[17:23:49.734]   SXR 01 : 8814d899y\n

[17:23:49.734]   SXR 01 : 881414e7y\n

[17:23:49.734]   SXR 01 : 8814d87fy\n

[17:23:49.734]   SXR 01 : 880715f3y\n

[17:23:49.734]   SXR 01 : 88184a67y\n

[17:23:49.734]   SXR 01 : 88184a59y\n

[17:23:49.827]   SXR 01 : 820101d5y\n

```

PostconEssen, 2019-01-24:
OK, i have a workarround for this:

.apn = "tm "

Just add one space after the "tm". 
Seems like there is some minimum length.

jniewerth, 2019-02-04:
Yes. That is exactly what I did as workaround. But I suspect that some carrier could perhaps not trim the apn name and then maybe it would fail.

jniewerth, 2019-02-04:
It seems to be a bug with when OS_Free is called on the internal pdp context structure. I tried to debug it but did not have the time yet to find the root cause. I think it is either a double free bug or some kind of off-by-one error when allocation the structure.

niwantha33, 2019-04-26:
rectified by adding space after apn

sebi5361, 2020-02-11:
@PostconEssen, @jniewerth, @niwantha33:
In his [post](https://github.com/pulkin/micropython/issues/28#issuecomment-584356753) @pulkin suspects that your network carrier doesn't check the APN name, and that you could enter what ever more-than-two-digit APN name and it would still work. Could you try with?
```
.apn = "test"
```
Thks

martinconstante, 2021-05-11:
I had the same problem and added the space and works!! Nobody understands that by one space we will become crazy!
Thanks a lot!!!

ekostero, 2024-02-23:
I have the same problem , probably the library 'cellular' for me at least with the function 'gprs' does not recognize less the 3 characters in APN value
Any help is welcome,
ps I tried space before and after with no luck