Ai-Thinker-Open/GPRS_C_SDK #265: compiler error when using asin or acos functions.

win32cpp, 2018-11-21:
Hi
SDK version : V2.112
When I use the following functions, I get a compiler error.
asin / acos

**undefined reference to `asin'**

lsaavedr, 2018-11-26:
`#include <math.h>` ?

lsaavedr, 2018-11-27:
seems to be a problem with cpp? you are compiling a cpp project?

lsaavedr, 2018-12-06:
the problem is with V2.112 (for example, this does not have many files [math.h] in the `libs/utils/src` folder), if you clone the git repository, the problem was solved:

```bash
git clone https://github.com/lsaavedr/GPRS_C_SDK.git
cd GPRS_C_SDK
git checkout update_iotkit-embedded
git submodule update --init --recursive
```

will88china, 2018-12-26:
@lsaavedr I found some anger value is not correctly calculated, for example: asin(0.625) should be 0.675132, but it has calculated to 1.570796, do you have any idea more? thanks.