Ai-Thinker-Open/GPRS_C_SDK #267: undefined reference to `sin'

alonomry, 2018-11-24:
## 1. SDK version（SDK 版本）

{
2.112
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

when i trying to calc sin/cos/sqrt functions i got this error "undefined reference to `sin' "
i tried to find a solution and add -lm to the makefile flags.
any other suggestions ?
thanks
}

---

lsaavedr, 2018-11-26:
`#include <math.h>` ?

alonomry, 2018-11-26:
Sure!

lsaavedr, 2018-11-26:
if you have `#include <math.h>` it is impossible that you have "undefined reference to `sin' ", can you post your code ?

alonomry, 2018-11-27:
I rean about it in some places and they said that its linking problem.
I have to add "-lm" while compiling it.
I tried add add it to makefile's flags without success..
Any other idea?

alonomry, 2018-11-27:
> print all errors... to see what happened...

I'll do it later cause I'm not with my computer...
Meanwhile you can see that post:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/169

lsaavedr, 2018-11-27:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/169 is closer, but this (https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/267) seems to be a problem with cpp? you are compiling a cpp project?

punee995, 2018-11-27:
I don't know if this going to help but there is an example for math function see [demo/std](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/tree/master/demo/std)

lsaavedr, 2018-11-27:
in cpp seems to be undefined `restrict`keyword, must be `#define restrict __restrict`

alonomry, 2018-11-29:
here is my code, it's calculating distnace between two cordinates(lat,lon):

double calcDistance(double A, double B, double C, double D) 
{
    double dLat;
    double dLon;
    dLat = (C - A);
    dLon = (D - B);
    dLat /= 57.29577951;
    dLon /= 57.29577951; 
    double v_a;
    double v_c;
    double distance;

    v_a = sin(dLat/2) * sin(dLat/2) + cos(A) * cos(C) * sin(dLon/2) * sin(dLon/2);
    v_c = 2 * atan2(sqrt(v_a),sqrt(1-v_a));
    distance = r * v_c;
    return distance;
}

and for example one of the errors that i got:
undefined reference to `sin'
value:0, p:ffffffff88241476, addend:0, symbol:0
relocation truncated to fit: R_MIPS16_26 against `sin'

alonomry, 2018-11-29:
> I compile without problem in a cpp project and in an c project your function, can you send a complete project?
> [projects.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2629272/projects.zip)

[https://github.com/alonomry/Hue/blob/master/dronetracker.rar](url)

lsaavedr, 2018-11-29:
```bash
[MAKE]  project/dronetracker

make[4]: Warning: File 'src/drone_tracker.c' has modification time 16479 s in the future
[CC]   drone_tracker.o             <== drone_tracker.c
src/drone_tracker.c: In function ‘gps_Task’:
src/drone_tracker.c:237: warning: unused variable ‘DistanceBetweenCoordinates’
src/drone_tracker.c:196: warning: ‘isFixedStr’ may be used uninitialized in this function
make[4]: warning:  Clock skew detected.  Your build may be incomplete.
[AR]   libdronetracker_debug.a
 

-----------------------------------------------
[LDGEN]             >>> cust.ld

[LD]                dronetracker.elf
[ElfCombine]        Elf binary & map file

[GEN]               strip platform symbol
                    |
                     --/home/lsaavedr/Nextcloud/dev/A9G/software/GPRS_C_SDK/platform/compilation/platform_symbols_to_strip

[SREC]              dronetracker.srec for flash/romulator
[LOD]               /home/lsaavedr/Nextcloud/dev/A9G/software/GPRS_C_SDK/hex/dronetracker/dronetracker_flash.lod
[LODCOMBINE]        Combine with Platform lod
[LODCOMBINE]        Combine sucessful

====== Build Time: 3s complete at 2018-11-29 12:14:48 =======
ROM total: 0x00100000(1048576) Bytes, used: 13744 Bytes (1.31%)
RAM total: 0x00100000(1048576) Bytes, used: 5184 Bytes (0.49%)
=============================================================
```

alonomry, 2018-11-29:
so what i'm doing wrong?
when you are compiling you are writing :".\build.bat dronetracker" ?
i have the last csdk , i tried to download it again today..

will88china, 2018-12-03:
我也碰到了这个问题，请问当前有解吗？

alonomry, 2018-12-03:
Not yet...
I still have the problem

lsaavedr, 2018-12-05:
I suspect that the problem is windows, now I will deal with windows, in ubuntu (linux) no problem (my windows is being updated)

lsaavedr, 2018-12-05:
i can confirm in windows:

```bash
build folder exist
number of processors: 4
Build host is WINDOWS

MAKE              init
CC                sdk_init.c
AR                libinit_debug.a

MAKE              libs

MAKE              libs/gps

MAKE              libs/gps/minmea
CC                minmea.c
AR                libminmea_debug.a

CC                gps.c
CC                gps_parse.c
PREPARING         libgps_debug.a

MAKE              libs/utils
CC                assert.c
CC                buffer.c
CC                time.c
CC                api_debug.c
AR                libutils_debug.a

PREPARING         liblibs_debug.a

MAKE              dronetracker
CC                drone_tracker.c
src/drone_tracker.c: In function 'gps_Task':
src/drone_tracker.c:237: warning: unused variable 'DistanceBetweenCoordinates'
src/drone_tracker.c:196: warning: 'isFixedStr' may be used uninitialized in this function
AR                libdronetracker_debug.a

LDGEN  by csdk    cust.ld

LD                dronetracker.elf
C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/build/dronetracker/dronetracker/lib\libdronetracker_debug.a(drone_tracker.o): In function `calcDistance':
C:\Users\Luis\DOWNLO~1\A9G\SDK\GPRS_C_SDK\dronetracker/src/drone_tracker.c:115: undefined reference to `sqrt'
value:0, p:ffffffff8824148e, addend:0, symbol:0
C:\Users\Luis\DOWNLO~1\A9G\SDK\GPRS_C_SDK\dronetracker/src/drone_tracker.c:115: relocation truncated to fit: R_MIPS16_26 against `sqrt'
C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:722: recipe for target 'C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/build/dronetracker//dronetracker.elf' failed
make[2]: *** [C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/build/dronetracker//dronetracker.elf] Error 1
C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:718: recipe for target 'C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/build/dronetracker//dronetracker.srec' failed
make[1]: *** [C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/build/dronetracker//dronetracker.srec] Error 2
C:/Users/Luis/DOWNLO~1/A9G/SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2
-------------------------------------------------
ROM    total:1048576 Bytes     used:18864 Bytes
RAM    total:1048576 Bytes     used:5184 Bytes
=================================================
Start Time : 13:06:32,48
End   Time : 13:06:41,12
Build Time : 8.964s
=================================================
```

lsaavedr, 2018-12-05:
Apparently the problem is in the construction of the library "libutils_debug.a" in windows it is of size 97K while in linux it is of size 419K, I am going to review this ...

lsaavedr, 2018-12-06:
the problem is with V2.112 (for example, this does not have many files in the `libs/utils/src` folder), if you clone the git repository, the problem was solved:

```bash
git clone https://github.com/lsaavedr/GPRS_C_SDK.git
cd GPRS_C_SDK
git checkout update_iotkit-embedded
git submodule update --init --recursive
```

alonomry, 2018-12-06:
> the problem is with V2.112 (for example, this does not have many files in the `libs/utils/src` folder), if you clone the git repository, the problem was solved:
> 
> ```shell
> git clone https://github.com/lsaavedr/GPRS_C_SDK.git
> cd GPRS_C_SDK
> git checkout update_iotkit-embedded
> git submodule update --init
> 
> cd libs/aliyun/iotkit-embedded
> git rm --cached src/packages/Link-CMP
> cd ../../../
> 
> git submodule update --init --recursive
> ```

works great! thank you so much!

will88china, 2018-12-07:
@Saavedra,  thanks for your effort! great!!!

stevohstine, 2019-01-01:
Hey .. Am Stephen..  Am getting the same error...  "undefined reference to cos"  please help

cryarchy, 2019-01-01:
> the problem is with V2.112 (for example, this does not have many files in the `libs/utils/src` folder), if you clone the git repository, the problem was solved:
>
> ```shell
> git clone [...]

Thanks. This worked for me for a bunch of different `undefined reference to *` errors.

Na55im, 2020-01-17:
Hey, this didnt work for me. 

> the problem is with V2.112 (for example, this does not have many files in the `libs/utils/src` folder), if you clone the git repository, the problem was solved:
>
> ```shell
> git clone [...]

still getting this output:
System Version : WINDOWS
---------------------------------------------------

[MAKE]  init

[CC]   sdk_init.o                  <== sdk_init.c
[AR]   libinit_debug.a
 

[MAKE]  libs

[MAKE]  libs/gps

[MAKE]  libs/gps/minmea

[CC]   minmea.o                    <== minmea.c
[AR]   libminmea_debug.a
 

[CC]   gps.o                       <== gps.c
[CC]   gps_parse.o                 <== gps_parse.c
[AR]   libgps_debug.a
        |
         -- libminmea_debug.a
 

[MAKE]  libs/utils

[CC]   assert.o                    <== assert.c
[CC]   hal_gpio.o                  <== hal_gpio.c
[CC]   buffer.o                    <== buffer.c
[CC]   math.o                      <== math.c
[CC]   ntp.o                       <== ntp.c
[CC]   log2f.o                     <== log2f.c
[CC]   stdlib.o                    <== stdlib.c
[CC]   log2.o                      <== log2.c
[CC]   time.o                      <== time.c
[CC]   api_debug.o                 <== api_debug.c
[AR]   libutils_debug.a
 

[MAKE]  libs/cjson

[CC]   cJSON.o                     <== cJSON.c
[CC]   cJSON_Utils.o               <== cJSON_Utils.c
[AR]   libcjson_debug.a
 

[AR]   liblibs_debug.a
        |
         -- libgps_debug.a
        |
         -- libutils_debug.a
        |
         -- libcjson_debug.a
 

[MAKE]  cjmcu8118

[CC]   hdc1080.o                   <== hdc1080.c
[CC]   CJMCU8118.o                 <== CJMCU8118.c
src/hdc1080.c: In function 'hdc1080_get_deviceID':
src/hdc1080.c:10: warning: passing argument 5 of 'g_InterfaceVtbl->I2C_ReadMem' from incompatible pointer type
src/hdc1080.c:10: note: expected 'uint8_t *' but argument is of type 'struct <anonymous> *'
src/hdc1080.c:9: warning: unused variable 'hdcId'
[AR]   libcjmcu8118_debug.a
 

-----------------------------------------------
[LDGEN]             >>> cust.ld

[LD]                cjmcu8118.elf
C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118/cjmcu8118/lib\libcjmcu8118_debug.a(hdc1080.o): warning: multiple common of `hdc'
C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118/cjmcu8118/lib\libcjmcu8118_debug.a(CJMCU8118.o): warning: previous common is here
C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118/cjmcu8118/lib\libcjmcu8118_debug.a(CJMCU8118.o): In function `SecondTask':
C:\GPRS_C_SDK\GPRS_C_SDK\cjmcu8118/src/CJMCU8118.c:64: undefined reference to `hdc1080_temp'
value:0, p:ffffffff882401ca, addend:0, symbol:0
C:\GPRS_C_SDK\GPRS_C_SDK\cjmcu8118/src/CJMCU8118.c:64: relocation truncated to fit: R_MIPS16_26 against `hdc1080_temp'
C:/GPRS_C_SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:752: recipe for target 'C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118//cjmcu8118.elf' failed
make[2]: *** [C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118//cjmcu8118.elf] Error 1
C:/GPRS_C_SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:748: recipe for target 'C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118//cjmcu8118.srec' failed
make[1]: *** [C:/GPRS_C_SDK/GPRS_C_SDK/build/cjmcu8118//cjmcu8118.srec] Error 2
C:/GPRS_C_SDK/GPRS_C_SDK/platform/compilation/cust_rules.mk:395: recipe for target 'lod' failed
make: *** [lod] Error 2

I only added hdc1080_temp and got this error.
[CJMCU8118.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/4078905/CJMCU8118.zip)