Ai-Thinker-Open/GPRS_C_SDK #268: Fetched in submodule path 'libs/aliyun/iotkit-embedded', but it did not contain e6bd..

joshuacox, 2018-11-26:
```
git submodule update --init                                                                                                          !7500
error: Server does not allow request for unadvertised object e6bd3556039538ac504c005f0c31dbdd0e906658
Fetched in submodule path 'libs/aliyun/iotkit-embedded', but it did not contain e6bd3556039538ac504c005f0c31dbdd0e906658. Direct fetching of that commit failed.
```

## 1. SDK version（SDK 版本）

{
git log -n1
commit 2d493e48b1413736eb8decdea38cdd1461d8ab4b
Author: Neucrack <CZD666666@gmail.com>
Date:   Thu Nov 8 16:52:28 2018 +0800

    optimize mqtt demo
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive

git submodule update --init 

also fails with the same error as above
}

---

joshuacox, 2018-11-26:
merge #266 to master!

```
$ hub pr checkout 266
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
Unpacking objects: 100% (4/4), done.
From github.com:Ai-Thinker-Open/GPRS_C_SDK
 * [new ref]         refs/pull/266/head -> lsaavedr-update_iotkit-embedded
M	libs/aliyun/iotkit-embedded
M	libs/gps/minmea/src
M	libs/lvgl/src
M	platform/csdk
Switched to branch 'lsaavedr-update_iotkit-embedded'
[  8:31AM ]  [ thoth@thoth-GS70-stealth:~/git/GPRS_C_SDK(lsaavedr-update_iotkit-embedded✗)(ruby-2.2.4) ]
 $ git submodule update --init
Submodule path 'libs/aliyun/iotkit-embedded': checked out '26fe75c8788e7e265596164632bc052838dcd38c'
Submodule path 'libs/gps/minmea/src': checked out 'cd27e72970cd50b46e9c95c1c0076a49c1ceaf33'
Submodule path 'libs/lvgl/src': checked out '59464fa41beff11780219b1b2eaae0d855d8d8ec'
```

joshuacox, 2018-12-17:
this has not moved to master and remains an issue

petrynchyn, 2019-02-28:
I have problem: git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
```
...
error: no such remote ref e6bd3556039538ac504c005f0c31dbdd0e906658
Fetched in submodule path 'libs/aliyun/iotkit-embedded', but it did not contain e6bd3556039538ac504c005f0c31dbdd0e906658. Direct fetching of that commit failed.
```

mattheys, 2019-10-08:
For anyone not familiar with git enough to checkout a pull request try these steps if this hasn't been merged.

ONLY DO THIS IF YOU GET THE SAME ERROR AS OP

git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
cd GPRS_C_SDK
git fetch origin pull/266/head:fixbranch
git checkout fixbranch
git submodule update --init

Fingers crossed that should checkout the submodules for you and give you a complete sdk.

joshuacox, 2019-10-08:
Github has also written a wrapper for git that makes this a bit easier:

https://hub.github.com/

so the above becomes

```
hub pr checkout 266
```

That being said, you should understand what is going on in the manual method before relying on that shortcut.