Ai-Thinker-Open/GPRS_C_SDK #269: I have downloaded GPRS_C_SDK but trojan.generic detected by antivirus

kesmtk, 2018-11-29:
Hello, 
I use IObit malware fighter licensed Pro version
and file ZbxLodToBin.exe is reported as Trojan

github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/platform/compilation/ZbxLodToBin.exe

lsaavedr, 2018-11-29:
ClamAV does not detect it as a Trojan:

```bash
$ sudo freshclam
Thu Nov 29 10:01:40 2018 -> ClamAV update process started at Thu Nov 29 10:01:40 2018
Thu Nov 29 10:01:40 2018 -> main.cvd is up to date (version: 58, sigs: 4566249, f-level: 60, builder: sigmgr)
Thu Nov 29 10:01:40 2018 -> daily.cvd is up to date (version: 25163, sigs: 2164378, f-level: 63, builder: neo)
Thu Nov 29 10:01:40 2018 -> bytecode.cvd is up to date (version: 327, sigs: 91, f-level: 63, builder: neo)

$ clamscan ./platform/compilation/ZbxLodToBin.exe
./platform/compilation/ZbxLodToBin.exe: OK

----------- SCAN SUMMARY -----------
Known viruses: 6723618
Engine version: 0.100.2
Scanned directories: 0
Scanned files: 1
Infected files: 0
Data scanned: 2.12 MB
Data read: 2.04 MB (ratio 1.04:1)
Time: 13.756 sec (0 m 13 s)
```

lsaavedr, 2018-11-29:
You can check in https://www.virustotal.com also...

https://www.virustotal.com/#/file/a09bfab87b4cf523b7e96bb1db253a909a0598f04837e3be5ee22f3d9f265e25/detection

kesmtk, 2018-11-29:
Thanks, so I can just white it from quarantine and use.

closed 2018-11-29 by kesmtk