Ai-Thinker-Open/GPRS_C_SDK #273: build.bat app 编译报错

pglin66, 2018-12-06:
E:\BaiduNetdiskDownload\A9G\TEL>build.bat app
build folder exist
number of processors: 4
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:115: WARNING: No platform elf file at path: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/debug
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:128: WARNING: No platform lod file at path: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/debug
-- System Version : WINDOWS
---------------------------------------------------
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory

[MAKE]  init
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory

E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory
[AR]   libinit_debug.a

[MAKE]  libs
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory

[MAKE]  libs/gps
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory

[MAKE]  libs/gps/minmea
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory

E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:8: E:/BaiduNetdiskDownload/A9G/TEL/platform/csdk/memd.def: No such file or directory
[CPP]  %-27s <== %s\n minmea.asm minmea.S
mips-elf-gcc: src/minmea.S: No such file or directory
mips-elf-gcc: no input files
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:997: recipe for target 'E:/BaiduNetdiskDownload/A9G/TEL/build/app/libs/gps/minmea/obj/debug/minmea.o' failed
make[6]: *** [E:/BaiduNetdiskDownload/A9G/TEL/build/app/libs/gps/minmea/obj/debug/minmea.o] Error 1
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:834: recipe for target 'E:/BaiduNetdiskDownload/A9G/TEL/build/app/libs/gps/minmea/lib/libminmea_debug.a' failed
make[5]: *** [E:/BaiduNetdiskDownload/A9G/TEL/build/app/libs/gps/minmea/lib/libminmea_debug.a] Error 2
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[4]: *** [dependencies] Error 2
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[3]: *** [dependencies] Error 2
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:748: recipe for target 'E:/BaiduNetdiskDownload/A9G/TEL/build/app//app.srec' failed
make[1]: *** [E:/BaiduNetdiskDownload/A9G/TEL/build/app//app.srec] Error 2
E:/BaiduNetdiskDownload/A9G/TEL/platform/compilation/cust_rules.mk:395: recipe for target 'lod' failed
make: *** [lod] Error 2
grep: platform\csdk\memd.def: No such file or directory
grep: platform\csdk\memd.def: No such file or directory
grep: build\app\app.map: No such file or directory
grep: build\app\app.map: No such file or directory
grep: build\app\app.map: No such file or directory

lsaavedr, 2018-12-06:
Are you download the GPRS_C_SDK or cloned from github? seems to be a wrong cloned...

harryhcs, 2019-06-01:
I have the same problem, did git clone recusrive, also doing docker one, same problem

xfdr0805, 2019-07-13:
> I have the same problem, did git clone recusrive, also doing docker one, same problem

me too

medida, 2019-08-07:
If cloneing then some modules are not downloaded correctly

I have found that if you run the modules seperately into the folder then it all works OK

these are all the modules you need.....

[submodule "platform/csdk"]
	path = platform/csdk
	url = https://github.com/Ai-Thinker-Open/GPRS-C-SDK-LIB.git

[submodule "libs/gps/minmea/src"]
	path = libs/gps/minmea/src
	url = https://github.com/kosma/minmea.git

[submodule "libs/aliyun/iotkit-embedded"]
	path = libs/aliyun/iotkit-embedded
	url = https://github.com/aliyun/iotkit-embedded.git

[submodule "libs/lvgl/src"]
	path = libs/lvgl/src
	url = https://github.com/littlevgl/lvgl.git

cl17t, 2020-01-20:
> 

> If cloneing then some modules are not downloaded correctly
>
> I have found that if you run the modules seperately into the folder then it all works OK
>
> these are all the modules you need..... [...]

thanks @medida ! work for me