Ai-Thinker-Open/GPRS_C_SDK #278: mips-elf-gcc command not found

sangam358, 2018-12-15:
please help to solve this 
i am trying to build this on ubuntu 18.10

administrator@GSN-yoga:~/projects/GPRS_C_SDK$ sudo ./build.sh demo gpio
param number:2
compileMode:debug
compile project gpio
compile path /home/administrator/projects/GPRS_C_SDK/demo/gpio
core number:4
Build host is LINUX

MAKE              init
CC                sdk_init.c
make[3]: mips-elf-gcc: Command not found
make[3]: *** [/home/administrator/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:959: /home/administrator/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o] Error 127
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/home/administrator/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:975: dependencies] Error 2
make[1]: *** [/home/administrator/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:718: /home/administrator/projects/GPRS_C_SDK/build/gpio//gpio.srec] Error 2
make: *** [/home/administrator/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: lod] Error 2
=== Build Time: 0s at 2018-12-15 18:47:17 ===
!!!!!!!!!!!!!!!!!!!!
   BUILD FAILED
!!!!!!!!!!!!!!!!!!!!

lsaavedr, 2018-12-15:
Please see: https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation_linux.html

Seems to be that you do not have the CSDK (tool chain) or you do not have it installed...

dev-sangam, 2018-12-17:
i tried 2 & 3 time , using link given by you but still have problem

toboxo, 2018-12-18:
同样的问题，是在Ubutu18.10

Neutree, 2018-12-18:
just restart your terminal, maybe the environment not take effect

wavresangam, 2018-12-19:
no still have  problem

wavresangam, 2018-12-19:
@lsaavedr  is there os dependency for compiling code
which os your using and what is version of os 

thanks & regards

craigMull, 2018-12-22:
I have the same problem running on a fresh install of ubuntu 18.04. I have followed the Wiki to the letter. Not quite sure what I am doing wrong. 

Please help 
```
devuser@devuserVM:~/workspace/AI_thinker/projects/GPRS_C_SDK$ ./build.sh demo gpio
param number:2
compileMode:debug
compile project gpio
compile path /home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/demo/gpio
core number:4
Build host is LINUX

MAKE              init
CC                sdk_init.c
make[3]: mips-elf-gcc: Command not found
/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:958: recipe for target '/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o' failed
make[3]: *** [/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o] Error 127
make[3]: *** Waiting for unfinished jobs....
/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:975: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:718: recipe for target '/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/build/gpio//gpio.srec' failed
make[1]: *** [/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/build/gpio//gpio.srec] Error 2
/home/devuser/workspace/AI_thinker/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2
=== Build Time: 0s at 2018-12-23 00:42:52 ===
!!!!!!!!!!!!!!!!!!!!
   BUILD FAILED
!!!!!!!!!!!!!!!!!!!!

```