Ai-Thinker-Open/GPRS_C_SDK #280: cool host time out error

rahul-yadav3766, 2018-12-18:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

wavresangam, 2018-12-30:
which usb to ttl converter you are using ??
recommended is cp210x

punee995, 2018-12-31:
I've tested on CH340 also FT232 both work great!

LeonSerri, 2019-05-01:
Hi,

I have the same issue. When I try to download, by using cool watcher, the .lod file I receive 'CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError).

I am using CH340 chip for USB to TTL, and seems to works fine (tested using putty).

I have also tried to press the reset button on pudding a9g but the issue remain.

Could you please help me?

Thank you in advance

DomolinBolivia, 2021-03-04:
La solucion esta en presionar el boton turn off chip y luego restart chip desde la aplicacion CoolWatcher developer.

A mi me funciono.

Richelfo, 2023-02-23:
> La solucion esta en presionar el boton turn off chip y luego restart chip desde la aplicacion CoolWatcher developer.
> 
> A mi me funciono.

cuando hago eso recibo tambien errores.
Que cable has usado con el A9G y que sistema operativo usas ?

collins-kariuki, 2024-04-07:
@Richelfo @LeonSerri @DomolinBolivia Any progress on this?
I am also getting  `CHST_ReadInternal': Request to coolhost timed out ! (CoolHostTimeoutError)