Ai-Thinker-Open/GPRS_C_SDK #281: Missing 8955 profile in Coolwatcher for CSDTK42_Linux

craigMull, 2018-12-23:
## 1. SDK version（SDK 版本）

{
V2112
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

when launching coolwatcher from CSDTK/bin folder, there is no option to select the 8955 profile as described in wiki: [https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/burn-debug.html](url)

}

I am not sure how to move forward with this
---

ichanjz, 2019-01-15:
@craigMull 
Interested to know if the issue was solved eventually, and how? I tried coolwatcher in different distributions of linux already, including the docker environment suggested here: [https://github.com/Neutree/gprs_build](url) and always got stuck at the same point as you did. I managed to move a little bit further though, by copying the missing 'profiles/8955.cwp' (and many other missing files) from another folder CSDTK/cooltools. At the end, I jumped to the conclusion that coolwatcher is for Windows users ONLY. Poor linux users like you and I still have to wait.
Perhaps someone cares to share some more successful experience in CSDTK for linux here

sytabaresa, 2019-05-25:
There is some advance in this issue?? many Linux users needs to use A9G hardware...

edvm, 2020-06-09:
++1, it would be great to have CoolWatcher running on Gnu/Linux ! :partying_face:

sbluhm, 2020-11-22:
just copy the profile from the Windows version to bin/profiles and adapt the paths.

logan-dunbar, 2022-09-03:
Instead of running just `coolwatcher` (which seems to point to `/home/logan/CSDTK/bin/coolwatcher` in my case), if I run `/home/logan/CSDTK/cooltools/coolwatcher` it seems to load with the profiles available. Hope it helps!