Ai-Thinker-Open/GPRS_C_SDK #282: Is it possible to run micropython on the A9G?

Skorpeo, 2018-12-23:
is it possible to flash the A9G with micropython?  If so, are there any instructions or flash tool?

PostconEssen, 2019-01-18:
Look here: https://github.com/Neutree/micropython

Skorpeo, 2019-01-20:
looks cool but not sure how to flash and no documentation...

pergerk, 2019-01-20:
You can flash it using coolwatcher.

pulkin, 2019-01-25:
Unfortunately, it does not look like wireless communications are implemented in this port.

pulkin, 2019-02-03:
I took over the micropython port at https://github.com/pulkin/micropython/tree/master/ports/gprs_a9 . I ported GPS and the next thing on the list is the GSM module itself. Testers are welcome.

Skorpeo, 2019-02-03:
Awesome! Hats off to you, when I have some time to flash the board I will give it a shot

pm4r, 2019-02-05:
Great job. GPS+GPRS are what people brings to this hardware. I'm too busy right now, but am looking forward to try this as well in a month or so.

pulkin, 2019-03-23:
I started porting sockets and, in principle, the port should be already usable for something as simple as GPS tracking application. I am aware of several ways to make the module unresponsive so watchdog is absolutely necessary. Have fun (I will no longer advertise in this thread).
```python
import cellular as c
import usocket as sock
import time

print("Waiting network registration ...")
while not c.is_network_registered():
    time.sleep(1)

print("Delay 2 sec ...")
time.sleep(2)
print("Activating ...")
c.gprs_activate("internet", "", "")

print("Local IP:", sock.get_local_ip())
print("Resolving ...")
ip = sock.dns_resolve("portquiz.net")
print("IP:", ip)

s = sock.socket()
print("Connecting ...")
s.connect((ip, 80))
print("Sending ...")
print("Sent:", s.send("hi\n"))
print("Receiving ...")
print("Received:", s.recv(128))
```

arlucio, 2019-05-10:
@pulkin @Neutree Im trying for a few days to get Micropython working on my module, and just can't.

I followed everything, generated the hex from ports/gprs_a9 (build was sucessful), uploaded the .lod file (also with success) but than I can't connect to module, upload file or do anything. Tried to get any response using pyserial, rshell or ampy, and nothing. Already tried using Windows and Ubuntu and also with the Docker image.

Programming/uploading generated hex files in the normal way using C++ works normally. I just can't get micropython to work with this module. Can anyone please give me some light here?

pulkin, 2019-05-10:
There could be several options. First, wiring. Are you sure you connect _another_ USB2TTL to TX1/RX1?
[image]

arlucio, 2019-05-10:
Ok my fault, now it's working. 

Actually I just had the idea to change pins to TX1/RX1 as you proposed, some hours ago (would be great if you could put it on the readme for the new users). 

I tried it before and it didn't worked but now it works, no idea why. Anyways thanks for the help, and great work!

garudaonekh, 2021-08-28:
Hi,
After I insert SIM and restart, it hangs and can't run any python code. But the its GSM modem seems to work because I can make a call to the number on the chip.
If I donot insert SIM, I can run python code.

Please note that I connect the power thru the 5v USB.

funcoleto, 2024-06-16:
doc..

https://github.com/pulkin/micropython/tree/master/ports/gprs_a9#cellular