Ai-Thinker-Open/GPRS_C_SDK #287: A9G's firmware verification error when download finished.

yoyojacky, 2019-01-04:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

Ogauy, 2019-01-11:
Please use the official software

Ogauy, 2019-01-11:
https://pan.baidu.com/s/16ZIQ9YbZdcVz-zvpzEFXbA

NerdBishop, 2021-01-10:
anyone with a link to the flash tool? and the firmwire?

VermaAyush, 2023-04-26:
please give the link of new firmware, The official Ai thinker wesite https://docs.ai-thinker.com/_media/gprs/ai-thinker_a9_a9g_at_v02.02.20190915r.rar file is corrupted. if anyone have working downloaded firmware

dipinraj, 2023-10-12:
> please give the link of new firmware, The official Ai thinker wesite https://docs.ai-thinker.com/_media/gprs/ai-thinker_a9_a9g_at_v02.02.20190915r.rar file is corrupted. if anyone have working downloaded firmware

Same issue here ! Any idea, where I can get firmware?

dipinraj, 2023-10-12:
> please give the link of new firmware, The official Ai thinker wesite https://docs.ai-thinker.com/_media/gprs/ai-thinker_a9_a9g_at_v02.02.20190915r.rar file is corrupted. if anyone have working downloaded firmware

Try different browser.I got resolved that way !

yoyojacky, 2024-08-22:
problem resolved. thanks for tips.

closed 2024-08-22 by yoyojacky