Ai-Thinker-Open/GPRS_C_SDK #291: A9G Pudding Stop Working after running for a while.

cryarchy, 2019-01-09:


rodri16, 2020-04-15:
I have same problem, after a long period of working the module won't start again, it seems like no power arrives to A9G but 5V is applied

JazminFerre, 2021-06-27:
Hi, i have same problem. Could u solve it? Thanks

sistemasici, 2022-04-02:
Hi, i have the same problem. i have 10 A9G GPS trackers and some cads fail, no reason.

cryarchy, 2022-04-23:
It's been a while since I last worked with the A9G dev board. At the time when I was, I figured out that most of the weird failures of the device disappeared when I switched to a more capable voltage regulator. I cannot remember the exact component that I used but I would suggest you try fiddling with the power supplied to the device first.