Ai-Thinker-Open/GPRS_C_SDK #292: Where can I find the A9Gs RDA8955 datasheet in english?

cryarchy, 2019-01-09:
SDK_VERSION: V2.112

I am hoping to develop code to run on the A9G using [Rust](https://rust-lang.org). Maybe port FreeRTOS for use with Rust on the device.

Also, can the A9G module and this SDK be used in a **production environment**? I am working on a large project that may end up needing about **30K A9G modules**. The code encompasses both GPRS and GPS functionality as well as frequency sampling using the available GPIOs. Someone stop me if it is a mistake.

Lastly, can all the pins included in the `Power_Type_t` enum perform both input and output functions or is it only *pin 0 - 7*?
```c
typedef enum{
    POWER_TYPE_VPAD = 0,            // GPIO0  ~ GPIO7  and GPIO25 ~ GPIO36    2.8V   //always on
    POWER_TYPE_MMC,                 // GPIO8  ~ GPIO13                        1.9V
    POWER_TYPE_LCD,                 // GPIO14 ~ GPIO18                        1.9V
    POWER_TYPE_CAM,                 // GPIO19 ~ GPIO24                        1.9V
    POWER_TYPE_MAX
}Power_Type_t;
```

Ogauy, 2019-01-11:
'm sorry we don't have the A9Gs RDA8955 datasheet in english.You can run the example(./demo/gpio) to see which pin you can use.

lsaavedr, 2019-01-14:
do you have the rda8955 datashet in other language?

vbtaneja, 2019-01-26:
How is VIO Used ?
If I want to Connect DHT22 working at 3.3V, Can I Connect the Data Line to - say IO1 ?

Can I Connect VIO to 5 V and then use a 5V Interface ?

romatou18, 2020-05-09:
Yes having the A9G documentation in English would be great ;-)

@cryarchy it looks like the only alternative to the native C SDK, is micropython. Which given how much power the A9G CPU has, should make it quite awesome.