Ai-Thinker-Open/GPRS_C_SDK #294: AGPS

ju-sti, 2019-01-09:
## 1. SDK version（SDK 版本）

{
current git
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
I know that the LBS server only works in China.
Does AGPS (or the BRDC GPD file) only work in China, too?

I'm testing the code from gps_agps demo but I get a "agps fail" from the GPS_AGPS method. I replaced the lat/long parameters with my real location for test. In tracer I see some missing GPS messages from UART when GPS_AGPS is called, so something is happening but it does not work.

Thanks for your help!

}

---

punee995, 2019-01-10:
yes AGPS only works in china.

Ogauy, 2019-01-11:
AGPS need the LBS server

win32cpp, 2019-11-26:
I have a similar question above.
Does anyone have any information on creating a BRDC GPD file?

aerosales10, 2020-03-27:
I'm having the same issue. 
bool result = GPS_AGPS(latitude, longitude, 0, true);
This command fails and the trace outputs:
`[GPS][src/gps.c][GPS_AGPS][921]>>"set binary mode fail"`
#379