Ai-Thinker-Open/GPRS_C_SDK #301: A9 usb port for alphanumeric keypad?

ichanjz, 2019-01-25:
## 1. SDK version（SDK 版本）

{
V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Trying to figure out how do-able it is to connect a keypad (usb 1.1) to my A9 pudding board. The info I've read so far only says ithe A9 usb port is v1.1 (which matches my keypad) but never say whether it's a host port or device port.  Couldn't find any demo code in the SDK for its usage either. Could someone throw some light?

}

---

pulkin, 2019-01-25:
I have some serious doubts that the on-board USB port is for anything except powering up the device. You will need a [USB host controller](http://www.hobbytronics.co.uk/usb-host-board-v2?keyword=usb%20host) which, apart from other things,

> Converts a standard USB keyboard into a serial TTL device generating ASCII characters.

Then you will be able to attach it to UART to read keyboard inputs. Yes, it is expensive. A better option is to find a PS2 keyboard and [to use it directly with digital inputs.](https://playground.arduino.cc/Main/PS2Keyboard)

ichanjz, 2019-02-19:
@pulkin 
Thanks for your response. That cleared a lot of my doubts on the subject. Your suggestion of a PS2 keyboard/keypad could be a very good one although these PS2 devices are disappearing these day. I surely will have a look at the code your pointed me to. Thanks again.

closed 2019-02-19 by ichanjz