Ai-Thinker-Open/GPRS_C_SDK #303: `dofastpfXfer': The flash programmer has not sent his READY event! (FastPf::FlashProgrammerNotReady)

overgara2012, 2019-01-28:
## 1. SDK version（SDK 版本）

{

V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Fastpf V2 over Host
Loading the lod files: 
host_8955_flsh_spi32m_ramrun.lod
Using flash programmer: 
call_B2112_debug.lod
Ramrunning the flash programmer...

Resetting SPI flash...
Done.
Entered Host Monitor mode.

No boot sector struct exists. Skipping ...
Event sniffer (re)starting. (Connection: Event Sniffer (COM2))
 `dofastpfXfer': The flash programmer has not sent his READY event! (FastPf::FlashProgrammerNotReady)

./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 216 | `dofastpfXfer'
C:/Users/overg/Downloads/A9G/CSDTK42/cooltools/lib/ruby/1.8/benchmark.rb | 293 | `measure'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 214 | `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 718 | `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 788 | `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 83 | `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 112 | `fastpfGo'

--

--

--

}

---

ReleaseBug, 2019-01-29:
hi,Omar !
you just load the wrong file.( maybe...)
The order is reversed.
Loading the lod files:call_B2112_debug.lod
Using flash programmer:host_8955_flsh_spi32m_ramrun.lod

try again.

overgara2012, 2019-01-30:
ok, it works , thanks