Ai-Thinker-Open/GPRS_C_SDK #307: Coolwatcher persistent crash on Linux 4.9 Debian

digitaico, 2019-02-02:
1.  SDK V2.112
 Linux 4.9
Debian 9 - Stretch

2.  After building any given project,  say demo imei,  I succesfully launch both coolhost and coolwatcher.  However after filling lastcomport and clicking OK, coolwatcher crashes leaving this error message on Cli:

`sh: 1: ldconfig: not found 
./coolwatcher: symbol lookup error: /home/jea/software/CSDTK/cooltools/libcoolhost_library.so: undefined symbol: _ZN6QMutex12lockInternalEv
`
About the first and second lines, I've just added PATH in .bashrc.

Looking forward there is this issue [https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/227](url) where many actions are recommended and I've executed as well:
1) On .bashrc comment LD_LIBRARY_PATH settings and add them to .build.sh. Done
2) Change CSDTK42/lib to lib.tmp

After renaming lib folder,  a new issue appears: it's contents apperar as missing.

So what can be done?

lsaavedr, 2019-02-02:
* `undefined symbol: _ZN6QMutex12lockInternalEv` => QMutex library is missing
 * QMutex library has `qmutex.h` header
 * if you find in https://packages.ubuntu.com tha package that have `qmutex.h` header is `libqt4-dev`

can you try installing `libqt4-dev`:

```#!bash
sudo aptitude install libqt4-dev
```

regards, Luis

digitaico, 2019-02-02:
libqt4-dev installed and updated. Version 4:4.8.7+dfsg-11
However,  the same, coolwatcher crashes, and the output of `c++filt _ZN6QMutex14unlockInternalEv` is again `QMutex::unlockInternal()`.  Even after post install update.

pulkin, 2019-02-03:
I just removed (renamed) all Qt libraries bundled in `CSDTK/lib`. Others remove the entire `lib` folder but that prevents building examples on my setup. In the end, it looks like the following:
```bash
~ ls lib
gcc                           libgmpxx.la              _libQtGui.so
insight1.0                    libgmpxx.so.4            _libQtGui.so.4
itcl3.3                       libgmpxx.so.4.1.2        _libQtGui.so.4.7
itclConfig.sh                 libiberty.a              _libQtGui.so.4.7.4
itk3.3                        libmips-elf-sim.a        _libQtNetwork.so
iwidgets                      libmpfr.a                _libQtNetwork.so.4
iwidgets4.0.1                 libmpfr.la               _libQtNetwork.so.4.7
libcloog.a                    libmpfr.so.1             _libQtNetwork.so.4.7.4
libcloog.la                   libmpfr.so.1.2.2         _libQtScript.so
libcloog.so.0                 libppl.a                 _libQtScript.so.4
libcloog.so.0.0.0             libppl_c.a               _libQtScript.so.4.7
libcooldisassembly.so         libppl_c.la              _libQtScript.so.4.7.4
libcooldisassembly.so.1       libppl_c.so.2            _libQtSql.so
libcooldisassembly.so.1.0     libppl_c.so.2.1.0        _libQtSql.so.4
libcooldisassembly.so.1.0.0   libppl.la                _libQtSql.so.4.7
libcoolhost_library.so        libppl.so.7              _libQtSql.so.4.7.4
libcoolhost_library.so.1      libppl.so.7.1.0          _libQtXml.so
libcoolhost_library.so.1.0    libpwl.a                 _libQtXml.so.4
libcoolhost_library.so.1.0.0  libpwl.la                _libQtXml.so.4.7
libcoolruby.so                libpwl.so.4              _libQtXml.so.4.7.4
libcoolruby.so.1              libpwl.so.4.0.0          libromulib.so
libcoolruby.so.1.0            _libQt3Support.so        libromulib.so.1
libcoolruby.so.1.0.0          _libQt3Support.so.4      libromulib.so.1.0
libcoolxml_library.so         _libQt3Support.so.4.7    libromulib.so.1.0.0
libcoolxml_library.so.1       _libQt3Support.so.4.7.4  libtcl8.4.a
libcoolxml_library.so.1.0     _libQtCore.so            libtclstub8.4.a
libcoolxml_library.so.1.0.0   _libQtCore.so.4          libtk8.4.a
libc.so                       _libQtCore.so.4.7        libtkstub8.4.a
libgmp.a                      _libQtCore.so.4.7.4      ppl
libgmp.la                     _libQtDBus.so            tclConfig.sh
libgmp.so.3                   _libQtDBus.so.4          tkConfig.sh
libgmp.so.3.5.2               _libQtDBus.so.4.7
libgmpxx.a                    _libQtDBus.so.4.7.4
```

digitaico, 2019-02-04:
thanks @pulkin, I did as you recommended, now when building,  new broken dependencies arise.  Iḿ giving up.  For sure the linux version of SDK is bugged or incomplete.

digitaico, 2019-02-06:
I finally found that my OS updates delete the lib32z1 library.  So being a bad idea to mix both 32bit and 64bit libraries,  I think it's better to install the A9G SDK in a container or a VirtualBox or any virtualized environment.  My actual installation on Debian 9,  it's working having put on hold some 32bit packages.

closed 2019-02-06 by digitaico

Makimars, 2025-02-11:
Just gonna add more info I gathered while trying to solve this in 2025.

It works in a VM, should also work with a docker or smth like that.

I created an appimage from the VM which runs on modern distros. It's very crude but it works if you don't want to bother with gathering libraries or running a VM.

https://github.com/Makimars/GPRS_C_SDK_Linux_coolwatcher_appimage/releases/tag/appimage