Ai-Thinker-Open/GPRS_C_SDK #310: sh: 1: mips-elf-objdump.exe: not found , sh: 1: mips-elf-ld.exe: not found

overgara2012, 2019-02-11:
## 1. SDK version（SDK 版本）

{V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
when I try to compile any demo file:

omarv@omarv-msi:~/A9G/projects/GPRS_C_SDK$ ./build.sh demo imei
param number:2
compileMode:debug
compile project imei
compile path /home/omarv/A9G/projects/GPRS_C_SDK/demo/imei
core number:8
Build host is LINUX

MAKE              init
CC                sdk_init.c
AR                libinit_debug.a

MAKE              libs

MAKE              libs/gps

MAKE              libs/gps/minmea
CC                minmea.c
AR                libminmea_debug.a

CC                gps_parse.c
CC                gps.c
PREPARING         libgps_debug.a
!<arch>

MAKE              libs/utils
CC                api_debug.c
CC                time.c
CC                buffer.c
CC                assert.c
AR                libutils_debug.a

PREPARING         liblibs_debug.a

MAKE              demo/imei
!<arch>
CC                demo_imei.c
AR                libimei_debug.a

LDGEN  by csdk    cust.ld

LD                imei.elf
ElfCombine        Elf binary & map file

GEN               stripped (rm syms) elf file imei.elf
sh: 1: mips-elf-objdump.exe: not found
sh: 1: mips-elf-objdump.exe: not found
sh: 1: mips-elf-ld.exe: not found
Cannot merge elf

CP  		  AUDIO CFP FILE......
AUDIO CFG         No audio cfp file
SREC              imei.srec for flash/romulator

LODTOBIN  /home/omarv/A9G/projects/GPRS_C_SDK/hex/imei/imei_flash.lod
LODTOBIN          Sucessful
LODCOMBINE        Combine user lod with Platform lod
LODCOMBINE        Combine sucessful

=== Build Time: 2s at 2019-02-11 11:55:53 ===
awk: run time error: not enough arguments passed to printf("%.2f%
")
	FILENAME="" FNR=0 NR=0
awk: run time error: not enough arguments passed to printf("%.2f%
")
	FILENAME="" FNR=0 NR=0
ROM total: 0x00100000(1048576) Bytes, used: 480 Bytes (inf)
RAM total: 0x00100000(1048576) Bytes, used: 16 Bytes (inf)
omarv@omarv-msi:~/A9G/projects/GPRS_C_SDK$ 

}

---

Johhnzero, 2019-04-15:
the sam problem,mips-elf-objdump.exe: not found , sh: 1: mips-elf-ld.exe: not found ,have you solved？

kwincu, 2020-09-06:
I have the same problem, please help us?