Ai-Thinker-Open/GPRS_C_SDK #311: Debugging with CSDK

pulkin, 2019-02-19:
I have a problem debugging my [micropython port](https://github.com/pulkin/micropython). I am using the latest CSDK build (`coolwatcher` v2.01.03 Linux). I am following recommendations from [here](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/burn-debug.html).

I've built successfully the port and flashed it but now I need to debug a piece of code crashing my A9G. I was able to receive the crash backtrace in the Trace tool in `coolwatcher` but when running GDB launcher I am not able to connect to the GDB server. Nevertheless, the ELF file seems to load properly: I see symbols and their addresses. Resolving addresses from the stack trace I see in my events does not work (addresses are not known to gdb). I need help printing the backtrace.

psatya111, 2019-04-27:
@pulkin:- I am trying to flash te A9 gsm board but not able to do that. could you please help on this.Please let me know the wiring connection and flashing steps. my mail id is psatya111@gmail.com.I have followed the A9 gsm guide but no luck. please help me.

Johhnzero, 2019-06-13:
How do you track down crashes?Can you describe it more carefully?