Ai-Thinker-Open/GPRS_C_SDK #318: FFC ID

digitaico, 2019-03-15:
In order to operate a mobile device in my country, I'm required to ask an IT authority to ... authorize.  They ask me for FCC-ID.

I've found that as a manufacturer, AI-Thinker has the FCC code 2AHMR.  However didn't find the A9G FCC-ID.  
[FCC search engine here.](https://apps.fcc.gov/oetcf/eas/)
Question Do the A9G has an FCC ID code?,  what is it,  or when do you expect to get it.
If not,  is there an FCC-ID approved AI Thinker device with GSM and GPS capabilities,  
tHANKS

Johhnzero, 2019-06-13:
I'll find out if I have this certification FCC-ID. I'm sorry I didn't reply for a long time.

Johhnzero, 2019-06-13:
This certification is not yet available. If you need a larger quantity, you can contact our company directly.

digitaico, 2019-06-13:
Thnaks,  I hope you get the certification,  the product is really awesome.