Ai-Thinker-Open/GPRS_C_SDK #319: Put StorageId into event type of API_EVENT_ID_SMS_LIST_MESSAGE

lvdiev, 2019-03-25:
Currently, StorageId seems not included in event params. Please put it in so we can know where the message is storage, in order to delete the message or do something.
Thanks.

Johhnzero, 2019-06-13:
What specific event information do you want? Can you describe it in more detail?

lvdiev, 2019-06-14:
Does this project live again? 👍 

The scenario is that I want to clear all the SMS from both SIM and Flash.
At first, I call SMS_ListMessageRequst().
Then, an API_EVENT_ID_SMS_LIST_MESSAGE event is fired and I will get SMS_Message_Info_t from pEvent->pParam1.

What I expect is that SMS_Message_Info_t should contain StorageId so I can delete it easily by calling
SMS_DeleteMessage(). But now, it does not.

P/S:
From my point of view about this SDK's designing, I think you are re-using SMS_Message_Info_t over many places, including  SMS_ListMessageRequst(). But SMS_ListMessageRequst() requests to list all **saved** SMS, that means we should know where the SMS are saved, right?