Ai-Thinker-Open/GPRS_C_SDK #320: I2C Functions Problem.

win32cpp, 2019-03-25:
Do 'I2C_Transmit' and 'I2C_Receive' functions work correctly?
When I use them, error 14 is returned.

Johhnzero, 2019-06-13:
What's your version and which demo is it?

win32cpp, 2019-07-05:
i use the last version released (V2.112) .
try use DHT12:
char data[5];
I2C_ReadMem(I2C2,0xB8,0x00,1,&data,5,500);

HEMO5566, 2024-09-06:
i have the same problem here 
i try many things and finally I find out that the read memory i2c read and write memory functions has problem in the addresses part the API doesn't accept any addresses It deal only with one address 
the API has this problem I hope the SDK to be update and solve this problem 
[image]
[image]
when I read and write directly from eeprom It work well >> if I write all data then try to read it I get only the last data entered to eeprom that mean that no address update